PiNet/PiNet #7: Problem Booting off Raspberry Pi

archey347, 2015-03-24:
I have managed to successfully install the Server and put  a copy of the client onto a sd card so that the pi can boot off it.

It gets an ip address from my local dhcp server ok.

It then goes on to try and mount nfs, or something along those lines.
It comes up with an arror saying "mount: device or reasouce busy"

It repeats this for a long time.

Then it trys to start a script and comes up with an error "/scripts/init-bottom/ltsp: line27: panic: not found".
 It then goes into a busybox shell. But that fails with an error "/bin/sh: Can't Access tty".

Any help would be grateful. Thanks in advance. I will post any other info if requested

gbaman, 2015-03-25:
@archey347 You mention it tries to mount NFS, does it say that? NFS is actually broken in the most recent Raspbian kernels (out of my hands), but PiNet should automatically use NBD.
Line 27 is it trying to mount /proc, which must be failing.
I am assuming you have rebooted the server etc?
Could you also try the "Check cmdline.txt file" section on this page? http://pinet.org.uk/articles/troubleshooting/boot-issues.html

archey347, 2015-03-25:
@gbaman I have checked the cmdline.txt file and all of the information seems ok.

I have managed to mount off the same nfs server from another Linux machine using the paths in cmdline.txt.

I have done all of the usual stuff like rebooting the server etc.

In the Busybox shell, I managed to find out that the nfs directory was mounted at /root/rofs, so it is mounting ok, but for some reason the os thinks it hasn't.

gbaman, 2015-03-26:
The issue is the Raspberry Pi shouldn't be trying to mount with NFS, as there is issues with NFS in this kernel release. It should be using NBD. Can you confirm your cmdline.txt file has ```nbdroot=``` in it instead of ```nfsroot```?

archey347, 2015-03-26:
No, it says `nfsroot=` in cmdline.txt.

gbaman, 2015-03-26:
Ahh, I have found the problem. In some rare situations, the system can get a little confused on installation and think you want to use NFS (which is currently broken).
There is a scenario where the NBD flag could get set after the initial SD card image is built.

To fix it, run ```Update-SD``` from the menu and copy the newly generated files to the SD card. That should fix it.

archey347, 2015-03-27:
That sorts out the nfs problem and it now tries to boot using nbd.

Unfortunutly, another problem has occured.

After the DHCP bit, it says `block nbd0: Recieve Control Failed (result -32)`
Then it shuts down the socket and clears the queue.

gbaman, 2015-03-30:
Sorry to say, but this one has me and the LTSP dev team baffled.
Last idea, can you copy the result of 
```grep 'nbd_server' /var/log/syslog```
Or at least the last 10-20 lines after a failed boot up to something like [Pastebin](http://pastebin.com/) so we can have a look and see if it is throwing any errors?

trimdeck, 2015-03-30:
hi - i'm having the same error.

the pi ip address is assigned by dhcp ok..
then i get something like:

rootserver: 192.168.1.1 rootpath:
filename:
Negotiation: ..size = 512 mb
bs=1024, sz537133056 bytes
[7.566017] block nbd0: Receive control failed (result -32)
...
busybox
(initramfs)

result of grep 'nbd_server' /var/log/syslog:

... nbd_server [2282]: Spawned a child process
... nbd_server [2284]: Negotiation failed/8a: Requested export not found
... nbd_server [2284]: Exiting.
... nbd_server [2282]: Child exited with 1

I'm so excited about getting pinet up and running - hope you can help :)

gbaman, 2015-03-30:
Ok, after some digging, have an idea. Can I get the full contents of ```/etc/nbd-server/conf.d/ltsp_armhf.conf ``` from either of you?

trimdeck, 2015-03-30:
hi - in the folder
/etc/nbd-server/conf.d/

there is one file:
swap.conf

gbaman, 2015-03-30:
Then there is the issue!
Can you confirm if ```/opt/ltsp/images/armhf.img``` exists?

trimdeck, 2015-03-30:
yes it's there

gbaman, 2015-03-30:
Ok, can you try running ```sudo /usr/sbin/ltsp-update-image --config-nbd /opt/ltsp/armhf``` and see if that fixes it?
Once it completes compressing, can you also check if ```/etc/nbd-server/conf.d/ltsp_armhf.conf``` now exists?

trimdeck, 2015-03-30:
sorry, same result with pi

and still no /etc/nbd-server/conf.d/ltsp_armhf.conf 

not sure if it is relevant but after compressing there was the message:

cp: cannot stat '/tmp/tmpRwHPpuq2GY/boot/.': No such file or directory

gbaman, 2015-03-30:
Ok, think I have it. For some odd reason, LTSP didn't seem to generate the config file when it built the operating system. Try ```sudo ltsp-config nbd-server```

trimdeck, 2015-03-30:
okay so, 
sudo ltsp-config nbd-server 
has now created
/etc/nbd-server/conf.d/ltsp_armhf.conf

and this changed things a little

ip form dhcp okay

/scripts/local-premount

picture of output here:
https://plus.google.com/116574731249614733928/posts/1FNzkP5zj3t

gbaman, 2015-03-30:
At least we are getting somewhere..
What is the output of ```ls /opt/ltsp/armhf``` when run on the server?

trimdeck, 2015-03-31:
morning,

so the output of ls /opt/ltsp/armhf is:

bootfiles   debootstrap   var

archey347, 2015-03-31:
I had been having a few problems with my Ubuntu System, which had Pinet installed on it. I had to rebuild it all by reinstalling the os, whick meant having a clean install of Pinet.

Having the clean install seems to have fixed it, and I have had no problems so far... Fingers Crossed!

Thanks For all of your help

gbaman, 2015-03-31:
@trimdeck there should be a lot more in there. So that explains is, it failed to build the operating system for some odd reason. Can be caused by dodgy internet connection. Best answer is do a full Ubuntu reinstall and that should sort it.

closed 2015-03-31 by gbaman

trimdeck, 2015-03-31:
this is a long shot - but I have the same error on two machines - one an existing ubuntu install and the other a fresh one especially for pinet... but they are both 32 bit not 64 - could this have anything to do with it?

gbaman, 2015-03-31:
Weird.. I don't usually test with 32 bit, but there is no reason it should cause any issues? Can you try a full fresh reinstall of Ubuntu and PiNet?

trimdeck, 2015-04-01:
hi @gbaman - so I installed ubuntu 14.04 64 bit from minimal cd - text based installer - as i'm using a machine with only 512 mb RAM - and reinstalled pinet - and everything is working great now. Thanks