PiNet/PiNet #10: PiNet on Raspberry pi 2 - sd card created on lubuntu 14.10 server hangs in boot...

Rukem1, 2015-04-07:
I have installed the PiNet server on a lubuntu which installed fine. Then I created an sd card with the files in my PiBoot directory. When I place the sd in my RPi 2, right after mounting root filesystem, I start seeing repeating messages of
:
IP-Config: eth0 hardware address b8:27:eb:7f:1e:f8 mtu 1500 DHCP RARP
IP-Config: No response after 4 seconds giving up.         # incrementing seconds count

note: the mac (hardware) address listed is the mac of my RPI 2. I can do ifconfig -a in the Busybox shell and verify that. Is it trying to connect to localhost?

I checked the cmdline.txt file to verify my server address. I have checked to see that ltsp_armhf.conf exists with 4 lines in it. I have done "ls /opt/ltsp/armhf" and see a complete listing of all linux directories.

Suggestions?

Rukem1, 2015-04-07:
Follow-up. I have tried this with 2 different RPI2's using both wired and wireless ethernet.
After root mount, begins running script /scripts/local-top and reports that eth0 is up.

gbaman, 2015-04-07:
@Rukem1 that seems to be suggesting your home router isn't replying with an IP address over DHCP. Are you sure it is correctly giving out addresses? To test, from memory you should be able to feed the kernel a manual IP address by putting ```ip=1.1.1.1```, replacing 1.1.1.1 with a spare address on your range into the cmdline.txt file on the SD card. Make sure to add it to the end of the line after a space as the cmdline.txt file can be a little picky.

Rukem1, 2015-04-07:
My understanding was that the ip in the cmdline.txt was the ip of the server. I have reserved an address on my router for the server. A standard os on the sd card has no trouble getting an ip from the router. Is there something different with the PiBoot sd?

Rukem1, 2015-04-07:
Sorry for all of the 'comments', but I have been searching around for answers and came across a link that mentions getting the same error as I am. It was mentioned as "a bug on ipconfig not implementing the DHCP standard correctly." This was a year ago, but I believe they were referring to ubuntu 12.04. I mentioned that I have lubuntu (same kernel, but lighter weight overall) 14.10. Perhaps that is a problem and I should go the 14.04 since it is the Long Term release version. Do you have any experience using PiNet with 14.10?

http://askubuntu.com/questions/426384/pxe-boot-recent-ubuntu-live-cds

gbaman, 2015-04-07:
That sounds like a viable possibility. Officially PiNet is supported (and tested) on Ubuntu desktop 14.04 LTS. I sadly don't have the time or resources to try and test every other distribution out there.
So that sounds like the best thing to try now. If you can get it working on Ubuntu 14.04, then you know it is something related to the version of lubuntu.

Rukem1, 2015-04-08:
OK. I spent half of the day up(down)dating to lubuntu 14.04.1 and I appear to still get the same results. I have a screenshot, if there is anything in it that you can see causing my problem.
[image] *This is from the RPi2, of course. I was hoping that this is something that someone had reported before.

I suppose I could try the standard ubuntu 14.04 but I believe the only difference from my lununtu is the GUI windows manager and some of the other features like abiword rather than libreoffice, etc.

gbaman, 2015-04-08:
Ok, seeing the screenshot helps a little. At that point, the Raspberry Pi has not even tried to contact the server, so software on the server shouldn't have anything to do with it. Can I just double check, you are plugging the Raspberry Pi into your home network? As opposed to running a cable directly from the server to the Raspberry Pi?
Also, did you try the suggestion earlier about putting ```ip=``` into the cmdline.txt file? That should mean it will skip DHCP and manually assign the Raspberry Pi the address, e.g. ```ip=1.1.1.1``` would assign the Raspberry Pi an IP address of 1.1.1.1.

Rukem1, 2015-04-08:
Here is the cmdline.txt after I inserted the ip into it.
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 init=/sbin/init-ltsp nbdroot=192.168.0.52:/opt/ltsp/armhf root=/dev/nbd0 ip=192.168.0.53 elevator=deadline rootwait
    ? IS there any special place I should place the ip address?
Screenshot with the new cmdline.txt:
[image]

gbaman, 2015-04-08:
Yeah, that should have worked? Maybe try this one
```dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 ip=192.168.0.53 init=/sbin/init-ltsp nbdroot=192.168.0.52:/opt/ltsp/armhf root=/dev/nbd0 elevator=deadline rootwait```
I have a funny feeling it won't make any difference, but worth a try.

Can I also check, what type of Raspberry Pi are you using?
- Pi1, model B revision 1
- Pi1, model B revision 2
- Pi1, model B+
- Pi2, model B

And do you happen to have a second Raspberry Pi you could test with? Also have you tried a second card or completely formatting the SD card and reloading on the boot files?

Rukem1, 2015-04-08:
Nope. Didn't make any difference.
Pi2 (model B, I guess)

I do have a second Pi2 and tried it with same results.
sigh

gbaman, 2015-04-08:
Try formatting the SD card and reloading the files on.
Also, can I confirm your network is setup like this?
[image]

Rukem1, 2015-04-09:
Yes, the network is setup as in your picture.
I will reformat and reload the SD card.
Thanks

Rukem1, 2015-04-09:
I reformatted the SD card using the raspberry pi and 'gparted.' Formatted with fat32.
 (Just for kicks, I formatted ext4, knowing it wouldn't work.It didn't even recognize the SD) 
Same results.

I know you have so many using PiNet successfully, I really appreciate your time. I brought up Epoptes (on the server) and it looked very interesting. Would love to see it in operation.

Rukem1, 2015-04-10:
Well. I installed the Desktop version of 14.04, installed PiNet and created an SD card for the RPi2.
After all of that, I still see the same (exact) results on startup. (See Above)
I'm going to try something else. I have the server and the RPi connected to the network through a 5 port switch to the router. 
Tomorrow, I will connect both to the router without the switch and see what happens. Everything else works fine through the switch, so I am running out of ideas.   ?????

gbaman, 2015-04-10:
I didn't think a fresh server install would help much. It will be something to do with the networking.
Best thing you could try is with a different router, different ethernet cable etc.
I have an idea which I will try out tonight which might help. Will let you know.

Rukem1, 2015-04-10:
I connected each device (server/RPi) to it's own port on the router and bypassed the switch. I expected that it would finally solve the problem, but it didn't. Tried both letting it get it's address from the router and assigning an ip in cmdline.txt with the exact same results.
I don't know if you are familiar with Comcast Cable, but it is their router that I use. I'm assuming you are in GB?

Wait, just thinking. My Comcast router has only 1 port so I have to up-link it to an 8 port switch to get the ports I need for my home. So I guess technically, I am always going through a switch, but with all of the wireless and wired connections in my home, I have never experienced an issue like this. I expect most schools and other installations would have to use a multiport switch also.    ???????????

Rukem1, 2015-04-12:
Question. I can't find a syslog or messages file. Is there somewhere where I can see the messages that flow by during the startup?

Rukem1, 2015-04-15:
Hi Andrew, Did you have any luck with the idea you were considering a few days ago?
I have been trying different combinations, but still get the exact same results. I even tried your 'bleeding edge' installation and didn't install any additional software with no success. 
I haven't tried a different router, but if that becomes a last resort, I can buy an upgraded router for my Comcast cable service. I get all of my wired and wireless internet through that router, so taking it down for any period would not be feasible (Rebellion.) 
I still find it hard to believe a router would be the problem when I use a fixed (cmdline.txt) ip.

Thanks again, Kem

gbaman, 2015-04-15:
Sadly the idea I had didn't turn out to work, so am back to square 1. I have another idea, but it is taking a little longer to find the time to write it.

Rukem1, 2015-04-15:
Ok. I appreciate that you have a LOT of things going on. I will wait until you have the time to respond.

gbaman, 2015-04-15:
Ok, here is an idea. Why not try setting up a DHCP server on your PiNet server? Keep in mind, when the DHCP server is running, you shouldn't connect it too your normal home network, will cause issues. But with a DHCP server on the PiNet server machine, you can plug it directly into the Pi and see if it boots.
Is worth giving a go, means can skip it getting another router.
https://help.ubuntu.com/community/isc-dhcp-server

Rukem1, 2015-04-16:
Thanks. I will try that soon and let you know.

gbaman, 2015-04-21:
Any luck with this?

Rukem1, 2015-04-21:
Hi Andrew. I am actually learning (re-learning) a lot while trying to set up my ubuntu server as a dhcp host. I'm trying to set it up to work with my ubuntu laptop before a pinet RPi would be a valid test. Not sure if dhcp will work over just a crossover cable, but I also tried connecting them with a switch. Not successful with the dhcp connection yet, so, No luck yet. It's been a very worthwhile experience so far though. I'm still working on that and other ideas.
BTW, in an earlier post I asked if there is message logging occurring on the PiNet SD card during boot up. Do you have an answer for that question?
Thanks for keeping in touch.

Rukem1, 2015-05-01:
Still working on it. Any idea on bootup logging file location?

gbaman, 2015-05-01:
I am not sure where it would log to. My best guess would be ```/var/log/syslog```, most stuff ends up in there.
Don't worry about crossover cable issues, the Raspberry Pi can auto switch internally so can use any cable.
Can you try something else for me? Here is an old set of SD card files. It is based on the 3.12 kernel branch which will not work with the Raspberry Pi 2, but should work fine with the Raspberry Pi 1 model B or Raspberry Pi 1 model B+.
Can you try booting with it as an experiment?
Of course, remember to add your server IP address in the cmdline.txt file.
https://dl.dropboxusercontent.com/u/1122948/temp/PiNet-3.12-rpi-boot.zip

Rukem1, 2015-05-02:
Thanks for trying Andrew, but I get exactly the same results with your 3.12 kernel copied to the SD (on my old B model.) There must be something different in my network, but I have NO dhcp problems with Linux, Windows, Android or iOS devices (more than  dozen) on my local network so it's a mystery to me. 
And using initramfs, there is no /var/log, or in fact most any of the regular Linux directories and files. I guess that's because it never gets far enough to download real Linux files from the server.

closed 2015-07-09 by gbaman

wewill147, 2016-01-08:
I'm wondering if there was ever a resolution to this problem. I am having the same issue.

gbaman, 2016-01-08:
William can you open a new issue fully describing the exact issue you are having and include the information listed on the support page?
http://pinet.org.uk/articles/support.html

wewill147, 2016-01-08:
Will do, thanks!