PiNet/PiNet #13: Make files localizable

ikbenmartijn, 2015-05-18:
We're currently doing a test setup in a school in Belgium. One of the main barriers we face is language. For instance: the 'handin' folder is a cool feature, but nearly unusable as a 8-yo doesn't recognize the word, he/she would understand 'indienen'.
It would be nice if we could change these strings in an option menu, but that would be messy and intensive and kind of 'the dumb approach'. The better way would be to provide the strings to be translated and 'outsource' the translations so .po files can be made and packed with the PiNet installation.
I'd love to help localizing PiNet (I can offer Dutch and French) but I'm going to need some help from your side as well :)

gbaman, 2015-05-18:
Having done a proper look into using GNU gettext to do this, it certainly looks viable to do it with strings in the software.
What is looking less viable though is files and folders like handin. The main reason for this being difficult is a file/folder is a bit more difficult to change after being created, especially if every single student has one in their home folder and say you have 300 students.
If I was adding something else new in like the handin folder, I would have to wait for each translation of the folder name before releasing the feature, which could take a number of months as it must be done before the release to anyone can be made. I also don't have the time or resources to be maintaining multiple branches, so for example if the German translator hadn't got back to me on it, holding back the update for German users. That can't be done.
I am open to suggestions though on finding a way around this issue as would love to offer internationalisation support.

ikbenmartijn, 2015-05-18:
So the issue actually falls apart in two separate issues:
1) Not a tech problem, rather a motivational problem: localization collection and timing (who will do it, will they keep doing it, how do we keep things going)
2) Tech problem: how do we localize already existing folders and files

I think n°2 is a serious issue, but one that can be tackled by:
- re-installing the server and clients
- educating users and motivate them to re-educate their users as well (basically: documentation/communication issue)

N°1 does not seem to have a real technical blocker, and however I understand you'd have to (want to) wait for ex. a german translator, the lack of german translators also suggests there's not a real need for a german translation.

I would suggest to start paving the road, allowing translations to be added (pull requests?) and see how things work out. I'm guessing the translation job for now is not a very hard and long job so people motivated to use this in their proper environment will surely make the effort, possibly bringing more users to the platform etc etc.

gbaman, 2015-05-18:
Well, I have given it a go adding the GetText identifier to each string and having it spit out strings.
They can be found here - https://github.com/PiNet/PiNet/tree/internationalisation/internationalisation
As you can see, there is a huge amount of strings (400+) that need translated in PiNet... And that is only the main script, there is also additional translations that would be needed to be done in the other scripts.

gbaman, 2015-05-18:
As a bit of a followup, I still need to figure out the best way to handle translations, most likely it will be a separate repository as it makes updates quite a bit easier. Will also needs to write a separate update system for translations on top of that.
Haven't made the decision yet on whether to go for it, but am certainly leaning towards going for it, just would need to build up a group of translators.

ikbenmartijn, 2015-05-18:
I think a separate repo makes very much sense.
I can't make the decision in your place unfortunately :) But just to give it a go, I've started translating some strings to Dutch. More to come in the next days and weeks :)

gbaman, 2015-05-19:
Ok, have done some playing around with the Microsoft translator API and wrote a Python script (mainly for the fun of it) to auto translate .po files into other languages. I though have no idea how far off the real thing it will be and given I only can speak/read English, I have no way of checking. On a scale of 1-10 (1 being completely unusable and 10 being perfect), how does this look? (Ignore the random /n's)
https://dl.dropboxusercontent.com/u/1122948/temp/PiNet-nl.po

closed 2015-07-09 by gbaman