PiNet/PiNet #18: Different problems

dlaboral, 2015-05-20:
Hi Andrew, 

My name is David and I the Technical Manager in a Art Center in Spain. We do different workshops for children, and we are very interested in Pinet. My coworker and I tried for three days to install Pinet in different Linux distribution but we always got different errors. 

First Problem

At the beginning when We excuted "sudo Bash Pinet" we got a screen "No internet connection" but we had internet connection (Telefonica) in our houses we got the same message. 

We test 14.04 LST Desktop and server 32 y 64 Bits, and 12.04 LST Desktop and server 32 y 64 Bits, and the only way to avoid the problem is using the mobile phone as router and them using the ADSL connection for the rest of the installation.  

Second Problem 

During the installation we see many times this errors: 

update-initramfs: Generating /boot/initrd.img-3.12-1-rpi
/bin/df: Warning: cannot read table of mounted file systems: No such file or directory
warning: failed to read mtab

cryptsetup: WARNING: could not determine root device from /etc/fstab
/bin/cat: /proc/cmdline: No such file or directory
/bin/cat: /proc/cmdline: No such file or directory

At the end everything looks fine, we copy the files into de SD card, Raspberry switch on It gets DHCP in the same network but It does not work. It is possible for the errors. Now we are installing the last Script 0.7 that you upload the last night, but we have the same problem with the alert "No internet connection", now It is installing.    

Thanks for your work and your help.

Best

dlaboral, 2015-05-20:
Hi Andrew again, 

We have the same problem with 1.0.7 version, testing with different ISP providers, different computers always we get "No internet connection, unable to proceed", also we use different DNS. Only with Ubuntu 14.02 connected throw the mobile phone as a router works sometimes, It is very strange.

gbaman, 2015-05-20:
Hey David,

The internet connection detector in PiNet can be a little over sensitive. It very much depends on your network and if your router is blocking stuff.
It will try to connect to 2 sites on the web via their direct IP addresses.
http://18.62.0.96
and 
http://74.125.228.100

If it fails to access both, then it decides there is no web connection. It can be a little funny though with certain network setups. I have added a new option in Release 1.0.8 (currently only in dev channel) to disable the internet checking. Be warned though... If you don't have a web connection, it could cause issues having the checking disabled.
To disable it (when running 1.0.8), open the PiNet config file ```sudo nano /etc/pinet```
and add onto the bottom ```CheckInternetConnection=false```.
PiNet should then ignore the state of the internet.

You can ignore those errors that you got during installation, those are normal and outside of my control.

Can you grab a picture of what you are seeing on bootup for the Raspberry Pis?
Also, have you worked through the bootup issues document yet? http://pinet.org.uk/articles/troubleshooting/boot-issues.html

closed 2015-05-22 by gbaman