PiNet/PiNet #19: How to manually create the files to the SD?

jcorreajr, 2015-05-22:
Hello

I'm trying to generate the SD files manually without downloading it from the git.

I made a copy of the kernel file (vmlinuz-3.18.0-trunk-rpi2) from /opt/ltsp/armhf/boot/, also copied the dtb modules and initrd.img-3.18.0-trunk-rpi2 file.
All are OK, except for the initrd.img-3.18.0-trunk-rpi2 file.
When I use the initrd.img-3.18.0-trunk-rpi2 file copied from the /opt/ltsp/armhf/boot/, an error occurs when starting the raspberry IP. "Could not load neither overlayfs nor aufs".
If I replace the initrd.img-3.18.0-trunk-rpi2 file with the file downloaded from git it starts smoothly.
I checked the kernel versions and I'm using are OK.

I tried to recreate the initrd with mkinitramfs command within the chroot, but the problem remains.
I also tried to create the initrd directly in Raspberry initiated by LTSP, but also experienced the same problem.

Is there a specific procedure that is done to generate the initrd made available on git ??

gbaman, 2015-05-22:
Just out of interest, why do you want to generate the SD card files manually? It is pretty simple to grab them off the github repository?

There is a bug in the version of LTSP that can be found in the current Debian Wheezy (and Raspbian) repos. The bug came about because OverlayFS was renamed to simply Overlay in the 3.18 kernel release and AUFS was dropped from the same release.
A patch is out, just not in the Debian/Raspbian repos yet.

The guide I wrote for myself is here - http://pinet.org.uk/articles/advanced/raspbian-kernels-process

Be aware though, I don't officially support customised boot files with PiNet, but you are than welcome to give it a go. The guide may be slightly out of date, so be warned.

jcorreajr, 2015-05-22:
I needed to understand why my generation initrd was not working.
But with your help could generate functional initrd. thank you

gbaman, 2015-05-22:
Great, let me know if you have any other questions.

closed 2015-05-22 by gbaman