PiNet/PiNet #22: Limitation on number of PiNet user accounts?

Musana-SCS, 2015-06-21:
In Kigo Prison in Uganda (1 Server + 14 Raspberry Pi 1B with Raspi-LTSP) they have created 140 user accounts over the past months.
The server suddenly does not allow them to create any new users.
This morning they have deleted some user accounts of released prisoners.
They have now succesfully created some new users, but these new users cannot login at the Raspberry's.

Unfortunately I am in the Netherlands now and communication with Uganda is difficult via Skype.
They are still running Raspi-LTSP as we are going to bring them 14 pieces Raspberry Pi 2B and the latest version of PiNet by 10 July 2015.

Question1: is there a limitation on number of PiNet user accounts?
Question2: if not, do you have any idea why they cannot create any new user accounts?
Question3: if yes, how can we increase the number of user accounts to say 2.000?

Please let us know what information you want us to provide.
Please note we have no internet in Kigo Prison, but hopefully we can relay screenshots using a USB-stick.

gbaman, 2015-06-21:
This is certainly an interesting one...
With PiNet tests, I can't seem to find an upper limit? I just added 1000 users using the [PiNet Workshop](https://github.com/PiNet/PiNet-Workshop) tool and it seems to work without any issue. I then tried manually adding another user through the normal PiNet software, again without issue.

I will say though, Raspi-LTSP used to always do some group checks on all its users each time the Raspi-LTSP control panel was opened. As I never tested with large sets of users back then, I didn't notice that this actually caused a very long wait if people had over 50 users.

This issue was fixed for the release of PiNet though so there no longer should be any issue.
That issue though was only really a cosmetic issue and should not have caused any issues when users went to try and log on?

One thing that has been suggested is perhaps the hard disk is full?
If you select the ```System-Status``` option in the Raspi-LTSP control panel, ```Free Space``` is one of the readouts.
That should be at least 5G. If it is any less than that, then that may be your issue.
[image]

Musana-SCS, 2015-07-04:
Unfortunately they had total power failure in Kigo Prison for many days, so no possibilities to dig into problem.
We hope to find some more money somewhere to install solar panels in the future...
Apologies for delay in answerring; finally we could communicate with prison staff today; Uganda - the Netherlands.
The good news is the inmates solved their problem themselves.
It appeared they had made an error in exporting user data, thereby corrupting the system.
They have put original files back and everything is working as usual.
Our apologies for any inconveniences caused.
As we are flying to Uganda end of this week we will not only install the latest version of PiNet but we ourselves will be in Uganda for six weeks to reinforce training of inmates and make the whole system more robust.
Thank you for your excelent work.

closed 2015-07-04 by Musana-SCS