PiNet/PiNet #33: failed install

trimdeck, 2015-08-19:
Hi,

I have tried to install pinet and hit some problems.

Every time I start up pinet it says there is an update (0.10.46), which I install and then update pinet as recommended.

But this happens every time I start up pinet. The same update.

After two times of going through update I just cancelled and went ahead and copied the boot files to an SD and tried starting up the Pi, but it is unable to connect to pinet and spits out same lernal panic message as if pinet server was not connected.

I tried doing the installation again through the option in the 'Other' menu. But had the same problem...

Any ideas?

Many thanks

trimdeck, 2015-08-19:
[image]
 I also get this at the start but that might just be because I am running pinet from the commend line - no gui?

gbaman, 2015-08-19:
It sounds very like you are for some odd reason running an ancient version?
The 0.10.x releases were back when the project was Raspi-LTSP.
It was only when the release numbers hit 1.x.x that the name was officially changed and the old Raspi-LTSP repository frozen.
0.10.46 is the most recent (and likely final update) Raspi-LTSP release.
This causes me to believe you may have by mistake installed Raspi-LTSP (which is not longer supported) instead of PiNet.

Your best path forward is do a full reinstall of Ubuntu (this is very important) and follow the official installation documentation on the PiNet site.
http://pinet.org.uk/articles/installation/getting_started.html

Let me know if that sorts the issue.

trimdeck, 2015-08-19:
right at the beginning we typed wget --content-disposition http://bit.ly/pinetbeta

but then had to download a python file - would that have anything to do with it?

trimdeck, 2015-08-20:
so I just cracked it! I did a full rebuild of the LTSP raspberry pi image and it just booted up!

From the main menu - PiNet shows it is 1.013 so that looks ok (which I think it was before as well??) not sure what was going on with the update message :/

Thanks - will enjoy using this :D

closed 2015-08-29 by gbaman