PiNet/PiNet #38: Custom "Client"-System

MTRNord, 2015-12-03:
Hi,
how can I make an Ubuntu for real Laptops that is working with PiNet? What do I need as the bootup Ubuntu (or Debian) and what has to be on the Ubuntu that is sended to the Laptops?

Is there an Doc about how to do it?

gbaman, 2015-12-03:
PiNet only supported ARM based clients, like the Raspberry Pi and does not have any mechanism included to support standard x86 machines, like normal desktops or laptops.
Although it would be possible, it is not something PiNet supports.

Are you planning to use Pis as well, or just laptops?

MTRNord, 2015-12-03:
My plan was to use PiNet as an backend for my homenet :) So I know that the existing PiNet images and boot up images only for the Pi but it would be nice if there is a way how I can make images like that for an x86 PC. Like which packages I need and what are configurations that are needed for PiNet it self

gbaman, 2015-12-03:
Your best bet is look into LTSP, the system PiNet is based off.
http://www.ltsp.org/

The easiest way to set up thin or fat clients for normal desktop or laptop machines is using this guide - https://help.ubuntu.com/community/UbuntuLTSP/ltsp-pnp

MTRNord, 2015-12-03:
ok so there are no more packages required for PiNet? can I use PiNet than nearly OutOfTheBox for my project? Of course I have to modify it I think.

closed 2015-12-10 by gbaman