PiNet/PiNet #40: BerryBoot

ezraholm50, 2016-01-15:
Hello, 

First of all thanks for this amazing piece of software! My question is if its able to add the PINET sd card files to the BerryBoot loader http://www.berryterminal.com/doku.php/berryboot.
Normally its just the 2nd os partition of other os's that needs to be mounted to BerryBoot.
With PINET I'm not able to figure out what files are needed. I hope someone can help me out!
Thanks again!

Kind regards,

Ezra

gbaman, 2016-01-15:
It would not be easy to add it with Berryboot, the reason for this is each set of boot files are different as the IP address of the server is inside the cmdline.txt file. Also, PiNet does not have a 2nd partition, only the boot partition with a custom config.txt, custom cmdline.txt and x2 initramfs's.

So to answer your question, given I have no idea how Berryboot works, I very much doubt it would work.

ezraholm50, 2016-01-15:
I see, I will mess around a bit and test it out. If I manage to fix it Ill
let you know. Thanks!!

closed 2016-01-17 by gbaman