PiNet/PiNet #43: Shutdown & Reboot not working

marc-odp, 2016-02-20:
To quit session on the client, there are 3 options in the dialog box : Shutdown - Reboot - Logout
Shutdown & Reboot don't do anything. Logout works.
So, to shutdown, the only option is to Logout and on the login screen to choose 'Preferences' on the bottom left. In that menu 'Shutdown' works.
Ubuntu 15.10 + Pinet 1.1.4

kusti8, 2016-03-06:
It looks like it's because sudo access is restricted?

marc-odp, 2016-03-06:
Thank you for your response. During install, I responded 'no' to the option 'enable the use of sudo for pupils. Used for GPIO access' because we don't plan to use GPIO. So you think this is a side effect of not activating that option ?
I don't have the possibility to try to change this at the moment (no access to computer). I will try when I have a possibility...

kusti8, 2016-03-06:
Yes, that's been my experience. I don't develop for Pinet, but that looks
how it works. But your students shouldn't need to shut it down. You should
actually be able to pull the power since it's all on the server.

marc-odp, 2016-03-06:
Your are right, this is certainly not a 'major' problem. However, it seems funny to teach students the proper (and safe) way to shutdown a computer on one side and on the other side to tell them to just remove the plug in this particular Pinet network configuration...

kusti8, 2016-03-06:
I agree. For now, just logout and then shutdown from there.

gbaman, 2016-08-21:
I am going to see about properly fixing this in PiNet Jessie, it should just mean giving users of the ```pupil``` group access to shutdown, reboot and halt.

gbaman, 2016-10-16:
This bug has now been fixed in the new Jessie development branch.
https://github.com/PiNet/PiNet/commit/c19445e91ff362e6fac9331476fb2223db0ca79a

closed 2016-10-16 by gbaman