PiNet/PiNet #46: Update to Jessie

kusti8, 2016-03-02:
I'm still very new to PiNet, but I've noticed that the script uses wheezy instead of jessie. Wheezy is no longer supported. Looking through the script it should be easy to just substitute all instances of wheezy to jessie and change the PGP keys. Anybody willing to look into this?

kusti8, 2016-03-02:
Never mind. I just noticed the jessiedev branch. I guess I'll be using that.

closed 2016-03-02 by kusti8

gbaman, 2016-03-02:
As much as I wish it was that simple, is very far from it!
I have been working on the Jessie build for a few months now and have been hit by issue after issue after issue.
Big issues like the new Sudoless GPIO not working, issues with Nuscratch and also the big one, the Raspbian OS really needs rebuilt from scratch.
So that means lots of supporting code is required to make that as seamless as possible.
Most organisations using PiNet would prefer stability over new features, so because of that, new stuff doesn't get released until it is super stable and tested.
Jessie code is already up on a separate branch. Right now there is only a single issue left that I can find and is with Sonic Pi not working. Sam Aaron, developer of Sonic Pi is on it, but he is a very busy guy.
So the Jessie release is now waiting till he gets a chance to take a stab at fixing it.
The code can be found here - https://github.com/PiNet/PiNet/tree/jessiedev
And if you want to try it, is info on how to use custom branches here - http://pinet.org.uk/articles/advanced/custom-branches-repositories.html

kusti8, 2016-03-02:
Thanks very much for the reply. I just saw that after I posted and found your twitter post commenting about how its almost done. There are some things that we are experimenting with and Chromium 48 is a must (only on jessie with custom install - chroot). Just looking at it for now, but it looks very promising. We probably won't be using Sonic Pi anyway.

gbaman, 2016-03-02:
Unfortunately unless it is in the apt repositories, I have no intention of adding custom programs like Chromium to PiNet. Makes it a pain to keep applications up to date etc. You should be able to do most stuff that was doable with Chromium, with Iceweasel (basically Firefox) which is now available from the Jessie apt repos.

kusti8, 2016-03-02:
I realize that and understand. I'll just install it with the method said on
the website, chroot. Thanks!