PiNet/PiNet #47: Ubuntu PiNet Server

ddezellem, 2016-03-30:
I am attempting to create a local network of pi computers using an old laptop with Ubuntu and a bunch of raspberry pi's.  Right now I have four set up with the Pinet software downloaded and the SD cards formatted.  Unfortunately when I check my system status for PiNet my Server IP address is blank.  I believe this is why I am getting the '/bin/sh: can't access tty; job control turned off' error message. 

Any ideas before I try full reboot?

gbaman, 2016-03-30:
What is the setup of your network? Is there a DHCP server on the network to hand out IP address (router usually)? Does the server itself have internet access?

Also, are you using a virtual installation (using virtualbox), or just installed directly onto the server computer hardware?

It may be possible the PiNet software simply wasn't able to detect the IP address of your server due to a wacky network setup. If that is the case, you can just manually edit the cmdline.txt file on the SD card to the address the server does have.
http://pinet.org.uk/articles/troubleshooting/boot-issues.html

ddezellem, 2016-03-30:
I have the wireless signal which is the four curved lines but I have wireless disabled. The connection information is under that icon but it is not lit up, I cannot click on it.

gbaman, 2016-03-30:
Then that means the server is not connected to the network. Can you double check the LEDs on the network port on the server is actually flashing? Perhaps try a different Ethernet cable or port on the router?

ddezellem, 2016-03-30:
Alright the router is connected now I have the arrows and the connection information lists an IPv4.  It has an IP address, a broadcast address, etc. 

>>> Andrew Mulholland <notifications@github.com> 03/30/16 3:17 PM >>>
Then that means the server is not connected to the network. Can you double check the LEDs on the network port on the server is actually flashing? Perhaps try a
different Ethernet cable or port on the router?

ddezellem, 2016-03-30:
When I ping the server from the raspberry Pi I get 3 packets transmitted, 0 packets received, 100% packet loss.
That tutorial you showed me did not solve that particular issue.

>>> Andrew Mulholland <notifications@github.com> 03/30/16 3:17 PM >>>
Then that means the server is not connected to the network. Can you double check the LEDs on the network port on the server is actually flashing? Perhaps try a
different Ethernet cable or port on the router?

gbaman, 2016-03-31:
Are the network lights flashing on the Raspberry Pi as well? Can you upload a screenshot of what happens when you run ```ifconfig``` on the initramfs shell on the Raspberry Pi after it fails to boot?

ddezellem, 2016-03-31:
Noting is flashing, the only reason I thought it was fixed was because the connection information became available.  Everything is hard wired through cat 5
cables, I even turned the wireless off on the router.  My Ethernet cables go as follows from computer to internet port on router, from port one on router to
giga port switch port 1, from ports 2, 3, 4, 5 to raspberry pi's...
Picture of screen after ifconfig is attached

>>> Andrew Mulholland <notifications@github.com> 03/30/16 5:01 PM >>>
Are the network lights flashing on the Raspberry Pi as well? Can you upload a screenshot of what happens when you run ifconfig on the initramfs shell on the
Raspberry Pi after it fails to boot?

gbaman, 2016-04-02:
No picture unfortunately came through (might be better to reply directly on Github so you can attach it?).   
If the lights on the Raspberry Pi aren't flashing, then that means there is no traffic going across the wire. This means most likely either a dodgy cable or dodgy network switch. Would it be possible to try it with a different network cable/switch?    
No lights flashing on the Pi end though makes it very unlikely it is a Pi issue. Can I double check what Raspberry Pi model you are using?

closed 2016-08-13 by gbaman