PiNet/PiNet #48: Scratch Sound Jessie

cwsutherland, 2016-04-14:
Sound works fine in all other apps but no sound in Scratch.

I just get the following

scratch
Executing: sudo -E /usr/bin/squeak /usr/share/scratch/NuScratch15012016.image 
execing: /usr/bin/../lib/squeak/5.0-3553/squeak  -vm-sound-alsa  /usr/share/scratch/NuScratch15012016.image 
sound_Start(default)
soundStart: snd_add_pcm_handler: Function not implemented
sound_Start(default)
soundStart: snd_add_pcm_handler: Function not implemented

I have seen this issue on other forums, and it is something to do with ALSA. I would love a fix. I have new classroom to setup and all the PI's are version 3. I guess I will need to use Jessie (warts and all) since these won't boot up from the Whezzy image, which are all version 2. Any advice would be appreciated, great solution. thanks.

gbaman, 2016-04-14:
Thanks for reporting this. After doing the first large scale Jessie test at the Northern Ireland Raspberry Jam this weekend, one of the kids also reported the same issue.   
It isn't the only application with sound issues either, Sonic Pi volume controls don't seem to work.
After a quick bit of digging, here is something to try. On a terminal on the server try
```sudo mv /opt/ltsp/armhf/etc/asound.conf /opt/ltsp/armhf/etc/asound_old.conf```
(You can always put it back with sudo mv /opt/ltsp/armhf/etc/asound_old.conf /opt/ltsp/armhf/etc/asound.conf```)
Then, open the PiNet control panel, go into **Other** and select the **NBD recompress**.

There is a chance that might fix it, although at the same time is a chance it could break sound in stuff like Sonic Pi. I will try to get a chance to get proper back to work on this within the next week or 2, but this is an idea you can give a go while you wait.

cwsutherland, 2016-04-14:
Thanks Andrew,

Well it didn't fix it and it didn't break it either, still the same error message, and yes I did remember to run the NBD recompress. However I don't have any issues with volume level as you described, all my pi's are connected over HDMI with sound coming through the monitor. Volume controls work as expected.

In fact the only issue I have is the Scratch sound one, everything else works perfectly. 

Regards

Craig
Date: Thu, 14 Apr 2016 01:30:31 -0700

cwsutherland, 2016-04-16:
OK Well I have got it working now.

All I did was 

sudo ltsp-chroot --arch armhf

then

apt-get purge scratch

Which deleted nuscratch and scratch

then 

apt-get install scratch

exit

ltsp-update-image /opt/ltsp/armhf

And we are away! Happy Days.

gbaman, 2016-04-16:
Well, that isn't really a proper fix for the problem, but will act as a stopgap fix for now.
Using the standard version of Scratch has the disadvantage of loosing all the new speed improvements and the builtin GPIO server. Will properly investigate when I get a chance, thanks.

cwsutherland, 2016-07-01:
OK, thought I would update this with the solution I have.

First, you need to

sudo ltsp-chroot --arch armhf

next install the Alsa oss support

sudo apt-get install alsa-oss

Now you have to edit the squeak script to use OSS sound.

sudo nano /etc/squeak

replace the instances of vm-sound-pulse with vm-sound-oss

finally edit /etc/modules. and add
snd_pcm_oss

to the file

perform a ltsp-update-image /opt/ltsp/armhf

After this all the scratch sounds work perfectly with nuscratch, it seems the newer versions of Squeak don't have vm-sound-pulse driver. The only problem now is the kids are spending way to much time playing with the interesting sounds instead of coding. I am going to remove a whole bunch of them for next semester.

Hope this post helps somebody.

timrowledge, 2016-07-01:
a) thanks for the recipe
b) have you tried the /usr/share/timitidyinstall.sh script to add the MIDI synth support yet?
c) pulse audio is on my list to try building sometime in order to ease working with the BT sound stuff recently added to Raspbian
d) don't hold your breath since unix sound stuff just makes my brain curl up and hide.

gbaman, 2016-10-16:
This seems to have now sorted itself out in the most recent versions of Raspbian software.

closed 2016-10-16 by gbaman