PiNet/PiNet #50: Uninstall PiNet on server?

ItsDaGeek, 2016-05-01:
I recently created a PiNet test environment, and now that it's in production, how do I uninstall the server software from the Ubuntu machine I originally installed it onto?

Thanks!

gbaman, 2016-05-01:
Unfortunately PiNet does not include an uninstaller. Given the many many configuration files it tweaks, packages it installs and other changes it makes, keeping track of all of these is very difficult.    
Although an uninstaller would be possible, it is very low on the priority list right now given a number of other pressing issues (Jessie release etc).    
This is why in red [at the top of the installation page](http://pinet.org.uk/articles/installation/installing-PiNet.html) it reminds users not to install PiNet onto a machine running other services as can't be easily removed.

Best bet is to reinstall the other machine again with fresh copy of Ubuntu.

ItsDaGeek, 2016-05-05:
Thanks for your reply.

We'll just have to note next time to setup a restore point before the install for future use.

closed 2016-05-05 by ItsDaGeek