PiNet/PiNet #51: Turtle module having issues in Python - Jessie Dev

cwsutherland, 2016-05-03:
Hello,

If I try to run import turtle from the TKinter library I just get the following error.

Python 2.7.9 (default, Mar  8 2015, 00:52:26) 
[GCC 4.9.2] on linux2
Type "copyright", "credits" or "license()" for more information.
>>> import turtle

Traceback (most recent call last):
  File "<pyshell#0>", line 1, in <module>
    import turtle
  File "turtle.py", line 2, in <module>
    # turtle.py: a Tkinter based turtle graphics module for Python
NameError: name 'shape' is not defined
>>> 

and

Python 3.4.2 (default, Oct 19 2014, 13:31:11) 
[GCC 4.9.1] on linux
Type "copyright", "credits" or "license()" for more information.
>>> import turtle
Traceback (most recent call last):
  File "<pyshell#0>", line 1, in <module>
    import turtle
ImportError: bad magic number in 'turtle': b'\x03\xf3\r\n'
>>> 

This works fine on a standard Raspbian image. I checked and I couldn't see any difference. I deleted the turtle.pyc file in the case of the 3.42 error, and rebuilt the ltsp image, but still no luck.

Would appreciate 10mins of your time to investigate, otherwise it's Wheezy roll back time for Saturday's class.

cheers

Craig

tjguk, 2016-05-03:
I'm not in front of a PiNet setup at the moment but...

... this looks like the kind of issue you get when you have a module in
the working directory called "turtle.py" -- typically the one you're
using to try something out in turtle.py.

Can you just confirm that this isn't the case, please?

cwsutherland, 2016-05-03:
Yes, I've had that before, kid's love to call their project 'turtle.py'

Here's what I have

root@ltsp145:~# find / -name "turtle.py"
/usr/lib/python2.7/lib-tk/turtle.py
/usr/lib/python3.4/turtle.py
/rofs/usr/lib/python2.7/lib-tk/turtle.py
/rofs/usr/lib/python3.4/turtle.py

tjguk, 2016-05-03:
On 03/05/2016 15:50, cwsutherland wrote:
> Yes, I've had that before, kid's love to call their project 'turtle.py'
> 
> Here's what I have
> 
> root@ltsp145:~# find / -name "turtle.py"
> /usr/lib/python2.7/lib-tk/turtle.py
> /usr/lib/python3.4/turtle.py
> /rofs/usr/lib/python2.7/lib-tk/turtle.py
> /rofs/usr/lib/python3.4/turtle.py

Thanks for being so thorough. Can you do the same for turtle.pyc,
please? (I know you said you'd removed one, but just in case...). It's
just so close to the usual symptoms, especially since the traceback is
pointing to a source line which clearly doesn't contain a reference to
"shape".

cwsutherland, 2016-05-03:
Yes, that was it found a turtle.pyc in the home directory. Thanks for your suggestion. 

cheers

closed 2016-05-03 by cwsutherland