PiNet/PiNet #52: Spell checker in Libre Office does not work (no en-gb dictionary)

mr-eggleton, 2016-05-09:
I have fixed locally by ...

```bash
sudo ltsp-chroot --arch armhf

apt-get install myspell-en-gb  mythes-en-us
apt-get remove hunspell-en-us

exit
ltsp-update-image /opt/ltsp/armhf

```

gbaman, 2016-08-14:
Thanks for reporting and including a solution. I have added it to most recent [development snapshot](https://github.com/PiNet/PiNet/commit/d20d98c41f2ed6b69933ecdc8f8ef62cac2d8017#diff-e9dbff4c7456910bfddfab7041ec798cR1503), hopefully to be properly released soon.
Will leave the issue open till is properly released and tested.

mr-eggleton, 2016-08-15:
This seems to be in a Jessie related branch. Is Jessie now used if you pick the standard   “Dev (development”) option rather than “Master (stable)”? or do we still need to use the instructions here http://pinet.org.uk/articles/advanced/custom-branches-repositories.html  I have a class that really needs Jessie in September.

This e-mail and any attachments that accompany it are intended only for the use of the addressee(s), and may contain information that is privileged, confidential or exempt from disclosure. If the reader is not an intended recipient, any disclosure, distribution or other use of this e-mail is prohibited. If you have received this e-mail in error, please delete it from your system and notify the sender immediately. Any views or opinions presented do not necessarily represent those of The Sheffield UTC Academy Trust.

The Sheffield UTC Academy Trust, a charitable company limited by guarantee registered in England and Wales number 07652696. Registered office address: 111 Matilda St, Sheffield, S1 4QF. VAT number 143239036. The Sheffield UTC Academy Trust is an exempt charity.

gbaman, 2016-08-18:
Jessie development is still being kept separate from the Wheezy branches for now. Haven't had much time to work on it over past few months and think most would prefer stability over new features, especially in schools. So stable and development channels in the PiNet software still point to Wheezy releases.
The WorkingOnJessie branch is simply that, for me to upload code am working on and test. It isn't for general use and the instructions in the custom-branches documentation won't work for it, given each upload doesn't have a separate release version. They will eventually be squashed (using new Github feature) into a single release when Jessie is looking more usable.

closed 2018-05-19 by gbaman