PiNet/PiNet #53: Python/Minecraft seminar on Pinet

busla, 2016-08-06:
Not sure if it´s appropriate to add this as an issue, as it is more a question from existing teachers/users. Please advise on where to add a community related question like this.

I am excited to hear about experiences teaching Python using Minecraft-Pi with 20 - 30 students. What hardware specs does your server have and how does it handle that kind of load?

Cheers,

gbaman, 2016-08-07:
@busla there is no official minimum hardware specs for the server, but it is worth keeping in mind all the actual processing is done on the Pis themselves for something like Minecraft Pi.
I usually suggest a machine preferably with 4GB of RAM and at least duel core from past 5 years, but is far more important your server has gigabit ethernet (most machines sold in past 7-8 years with ethernet do) and have decent networking in the classroom. A gigabit network switch is recommended, or a network switch with 100mbit/s ports and at least 1 gigabit "stacking" ethernet port to act as the server port. Stay clear of networking equipment which is entirely 100mbit/s.

Hopefully that answers your question. As for a place for community questions, it is usually the Github issues. Other avenues have been explored without any fruit so far.

busla, 2016-08-08:
Thanks for the information. 

We are looking into PiNet as an option in for a CS course in an elementary school in Iceland. This will solve SO many issues :-)

Another question I have is about the GPIO. Our plan is to have a IOT course so I want to understand the communication between the Pi clients and the server in terms of GPIO. You have probably written about this so a URL might do, unless you feel like sharing it here :-)

gbaman, 2016-08-08:
@busla with PiNet, the Pis are simply like normal Pis running Raspbian for a majority of things, this includes GPIO. Because the Pis are set up as "Fat Clients", it means all processing is done locally on the Pi itself, so it has access to all hardware like a normal Pi would.

So GPIO is exactly the same as with usual Raspbian.

closed 2016-08-10 by gbaman