PiNet/PiNet #55: Printing support

gbaman, 2016-09-14:
Network printer support has been one that has been requested numerous times over the past few years for PiNet.
The Raspberry Pi doesn't by default ship with support with Raspbian for printing, but perhaps it may be doable with a CUPS server on the Ubuntu server which is then shared to the Pis?

## Todo
- [ ] Investigate CUPS for server.
- [ ] Investigate USB printer support (connected to the server).
- [ ] Investigate sharing network printer from server.

haxsaw, 2016-09-19:
Printing would be a great enhancement for PiNet; the teachers at our startup school really like the PiNet machines, but are a bit flummoxed by the lack of printing. I explained the focus on electronically handing in work, but in a lot of cases the kids have been assigned research that they need to take away, or else a teacher and student would like to review a piece of work together for a number of different reasons (our PiNet machines are only in our library; teachers may BYOD). In the interim, I'm creating a script that will run on the central server that will poll a "to_print" folder in each child's home directory, proceed to print out any files deposited there, and then remove the file afterward. This is our stop-gap measure to support basic printing; more complex jobs will be moved via USB stick to one of our Windows media studies machines. If there's any interest in my central printing script I'd be happy to contribute it.

gbaman, 2016-09-19:
Would be interested in if @tjguk or @waveform80 have any ideas on best way to implement this.
Unfortunately, with the many different printers out there, coming up with a way to do this nicely is no simple task unfortunately, even with CUPS (hence why I haven't already done it).

waveform80, 2016-09-20:
CUPS is your best bet, but as you've noted printer support is variable in Linux. As a general rule of thumb, if your printer is listed as working at http://openprinting.org/ then it should be pretty easy to set up (I use a CUPS server on Ubuntu to share an old Brother laser over the home network). Probably worth looking at the standard config of CUPS on Ubuntu; it's changed recently (during the 16.04 upgrade) and there's a ton of default security stuff in there I haven't had the time to dig into (it's effectively running an HTTP server as root, so it's worth being careful about the security).

It should be made abundantly clear that if the printer in question *isn't* listed (or listed as perfectly working) at openprinting.org then there's precious little we can do about it.

austinfarley, 2017-01-04:
I would like to help out with this where I can. Maybe I'm just not seeing it in the documentation but I'm unsure where the Pi grabs it's boot files from the server. I do have a use case for this that is not classroom based where It could be very helpful.

gbaman, 2017-03-08:
After some digging around on this one, turns out LTSP already has a decent chunk of support built in that can be piggy backed off.
It looks like it is as simple as installing CUPS on the Raspbian chroot, then configuring printers on the PiNet server and making sure they are set as shared.
The next PiNet Jessie release will include CUPS on Raspbian and documentation is in the works.

closed 2018-05-19 by gbaman