PiNet/PiNet #56: Move to bootcode.bin network boot

gbaman, 2016-09-15:
With the new Raspberry Pi 3 network booting support, a new bootcode.bin has been created by Raspberry Pi Foundation that allows older Pis to also network boot, with just the bootcode.bin on the SD card.
It solves a few issues including
- Out of date boot files. The bootcode.bin is a lot easier to update as just a single file.
- ProxyDHCP based booting, so the server is found automatically. No more to hard code the IP address of the server in cmdline.txt file.
- Far easier editing of boot files by the server admin on the fly. This makes https://github.com/PiNet/PiNet/issues/54 viable.
- Possible support for customised boot files per Pi (based on serial number).

## Todo
- [ ] Investigate support for different Pi versions
   - [x] Pi 1 B - Currently broken, bootcode.bin bug being investigated by Raspberry Pi Foundation.
   - [ ] Pi 1 B+ - Untested.
   - [x] Pi 2 B - Works as expected.
   - [x] Pi 3 B - Works as expected.

- [ ] Investigate booting using initramfs - Looking like is broken, bug being investigated by Raspberry Pi Foundation
- [ ] Figure out how to nicely pass IP address to LTSP boot scripts.
- [ ] Implement NBD booting support.
- [ ] Wait until new bootcode.bin is out of testing branch.

gbaman, 2016-12-05:
Unfortunately this one is still on hold as don't believe has been any movement yet on bootcode.bin support with initramfs. 
Won't be included in initial PiNet Jessie release.

@ghollingworth had any luck with it yet?

ghollingworth, 2016-12-05:
Still waiting for you to check if it works from an SD card...

gbaman, 2017-01-21:
Further investigation of this using bootcode.bin and accompanying files (from the ```next``` branch only currently) has it working with all Pis bar the Pi1B.
Initial implementation code for PiNet is ready to go live once Pi1B support is available and boot files are in the main branch.
@ghollingworth any chance of a rough ETA for Pi1B support for booting with bootcode.bin? Would be great to get this out as an option for PiNet users.

ghollingworth, 2017-02-09:
Pi 1B is now working on latest master branch of firmware

gbaman, 2017-02-09:
@ghollingworth fantastic! I will have a play over next few days (hopefully) with it.

ghollingworth, 2017-03-02:
Happy to close now