PiNet/PiNet #64: Multiple Custom config.txt files

Lewiscowles1986, 2016-10-24:
Read https://pinet.org.uk/articles/advanced/custom-config-file.html today. I Have an idea (not sure how good or bad it is).

Using ansible you can group computers to allow orchestration. The problem is you would have to know ahead of time which pi's go to which screens, then add their ip's to an inventory file. 

> This could be a PITA, although if you use the server as a DHCP server you can run scripts upon pi registration with a server, or write a script to discover for a dynamic inventory http://docs.ansible.com/ansible/intro_dynamic_inventory.html

You would need to setup to remember connected PC's using MAC address and long-lease times.

If you were to add ansible to the server installed dependencies #59 is likely a blocker as it uses SSH for the most-part. 

The reason I suggest ansible is that it uses python and SSH, so is quite lightweight and fits in with existing technologies. It can be used to define that specific nodes copy a specific `config.txt`, whilst other nodes receive a separate `config.txt` by assigning groups in an inventory file, or to write something akin to 
***Sample***
```ini
[servers]
1.1.1.1

[pi-net-vga]
10.0.0.11

[pi-net-hdmi]
10.0.0.5
10.0.0.6

[pi-net-tv]
10.0.0.20
```

 * http://docs.ansible.com/ansible/copy_module.html

gbaman, 2017-01-21:
As great a platform as ansible is, I am not keen on adding any additional major dependencies like that to PiNet to accomplish a small job like this. 
The intention is to allow the PiNet admin to edit a config file that is then appended onto the main PiNet config file for all devices. A majority of the code for this is already finished and is pegged for release as part of the .1 release for PiNet Jessie.
Right now there is no intention for PiNet to support multiple config.txt files for different Pis. The main reason is the complexity around it is significant, especially when it would only be used by a users.

closed 2017-01-21 by gbaman

TheFlyingCrocodile, 2017-08-09:
@gbaman i know this is closed but cant you simply add 3 config file option in the pinet menu for 1, 2, 3, then have it use the correct one based on processor kernel? useful for networks using different pi versions like a 2 set to 1000mhz is an overclock but having pi 3's also set to that is under clocking them. but im no expert