PiNet/PiNet #66: Enable sudo for students

Lewiscowles1986, 2016-10-24:
https://github.com/CODESIGN2/PiNet/blob/master/pinet#L981

you can `sudo chmod` or `sudo chroot` the entries needed every boot and add the user(s) to a group so that students don't need sudo access.

gbaman, 2016-10-25:
This is similar to what has been done for sudoless GPIO in Jessie with the new driver. It is a little more complicated though than has been documented there, given the fact that the chroot as it boots (and runs udev rules), the normal groups on the server don't actually exist with LTSP/PiNet)
The are added a little later in the boot process. Although it is possible to manually add a group to the chroot, that works most of the time until the 2 happen to have a different GID, at which point the mismatch causes it to give up and just set it as owned by root.

It is one I am finalising for the Jessie release now but given the new GPIO device wasn't ported back to Wheezy, this isn't going to be implemented into Wheezy for the current GPIO device (mainly for security reasons).

Lewiscowles1986, 2016-10-25:
could you setup a group on image build and then use that in the rules and add all users to that group?

gbaman, 2017-01-21:
With the PiNet Jessie Alpha, it is set up with a set of groups for accessing devices which are marked as the owner similar to how it is handled in Raspbian. As was the plan from the start, the code adds a new group to both the chroot and server with exactly the same group ID (and double checks this later down the line in case fiddled with). This is what allows the sudoless GPIO to work in PiNet Jessie.

closed 2017-01-21 by gbaman