PiNet/PiNet #70: IP address detection on server without English as primary language

gbaman, 2016-11-19:
Michael from Germany has stumbled across a great bug with current stable version of PiNet.
PiNet uses 
```bash
IP=ifconfig  | grep 'inet addr:'| grep -v '127.0.0.1' | cut -d: -f2 | awk '{ print $1}';
``` 
to work out the current IP address of the server. It then uses this when building the SD card image.
This code has been in PiNet for years now and works perfectly 95%+ times on servers which have their primary language set to English, hence why hasn't been changed since initially added.
But, as Michael discovered, if not using English, parsing the output of ifconfig doesn't work particularly well. For example, in German, "inet addr:" is actually "inet Adresse" which of course causes PiNet to be unable to detect the IP address.
It isn't a huge issue given PiNet doesn't actually use the IP address anywhere beyond top of main menu, status page and setting the address on the SD card image (which can also be done manually), but still a good idea to get it fixed.

Am going to look into seeing if is a better way to do this, suggestions welcome.

tjguk, 2016-11-20:
For networkzero I settled on using netifaces for this, although my needs 
were more complicated than yours appear to be:

   https://pypi.python.org/pypi/netifaces

Might be overkill for the task but...

TJG

gbaman, 2017-05-14:
This issue is now fixed and has been for a number of months. PiNet now uses the netifaces everywhere possible.

closed 2017-05-14 by gbaman