PiNet/PiNet #81: Minecraft Pi multiple worlds

gbaman, 2016-12-21:
With PiNet, right from the start Minecraft Pi has only supported a single world. When in the world selection section, any previously created worlds do not show. But, when you select "Create new", it just loads up the previous world anyway.

As for what is causing it, I am not sure... My hunch is it will be related to the fact the OS (outside of /home/user) is wiped on reboot, or that we are not using normal "Pi" user. It needs a thorough investigation though at some point.

gbaman, 2017-01-21:
Having looked into this further, am starting to wonder if it is a permissions error (along with a MCPI bug). When run with sudo, worlds appear as expected. This of course though isn't a solution given /root (where worlds are stored then) is wiped on reboot.
Still needs even further investigation.

cvincross, 2017-03-03:
I've been trying to figure this one out... can't seem to get to the bottom of it either.  So I came up with a work around.  I've made a script that copies the user's .minecraft directory into root and runs minecraft as sudo and then copies the folder back out to the student folder when minecraft terminates.  I incorporated this script into one that takes the user name and patches the minecraft binary so that they show up in multiplayer as the first 7 characters of their username instead of StevePi.  I've included the combined script in case it helps anybody else - it may be ugly but it works ;) (as long as you are wiling to give students sudo powers (which is fine for my setting)  (BTW...The minecraft name patch part can be done without sudo as long as you change the /opt/minecraft group ownership to 'pupil' in the base filesystem before compression.

#!/bin/bash

function patch_strings_in_file() {
    local FILE="$1"
    local PATTERN="$2"
    local REPLACEMENT="$3"

    # Find all unique strings in FILE that contain the pattern 
    STRINGS=$(strings ${FILE} | grep ${PATTERN} | sort -u -r)

    if [ "${STRINGS}" != "" ] ; then
        echo "File '${FILE}' contain strings with '${PATTERN}' in them:"

        for OLD_STRING in ${STRINGS} ; do
            # Create the new string with a simple bash-replacement
            NEW_STRING=${OLD_STRING//${PATTERN}/${REPLACEMENT}}

            # Create null terminated ASCII HEX representations of the strings
            OLD_STRING_HEX="$(echo -n ${OLD_STRING} | xxd -g 0 -u -ps -c 256)00"
            NEW_STRING_HEX="$(echo -n ${NEW_STRING} | xxd -g 0 -u -ps -c 256)00"

            if [ ${#NEW_STRING_HEX} -le ${#OLD_STRING_HEX} ] ; then
                # Pad the replacement string with null terminations so the
                # length matches the original string
                while [ ${#NEW_STRING_HEX} -lt ${#OLD_STRING_HEX} ] ; do
                    NEW_STRING_HEX="${NEW_STRING_HEX}00"
                done

                # Now, replace every occurrence of OLD_STRING with NEW_STRING 
                echo -n "Replacing ${OLD_STRING} with ${NEW_STRING}... "
                hexdump -ve '1/1 "%.2X"' ${FILE} | \
                sed "s/${OLD_STRING_HEX}/${NEW_STRING_HEX}/g" | \
                xxd -r -p > ${FILE}.tmp
                chmod --reference ${FILE} ${FILE}.tmp
                mv ${FILE}.tmp ${FILE}
                echo "Done!"
            else
                echo "New string '${NEW_STRING}' is longer than old" \
                     "string '${OLD_STRING}'. Skipping."
            fi
        done
    fi
}
# give student ability to write to minecraft binary to patch it
sudo chown -R root:pupil /opt/minecraft-pi

# getting username
long_name="$(whoami)"
# shortening username to 7 characters in order to patch successfully
name=${long_name:0:7}

patch_strings_in_file /opt/minecraft-pi/minecraft-pi "StevePi" $name

# Loading user files into root .minecraft directory
sudo \cp -r "/home/$long_name/.minecraft" /root/

sudo minecraft-pi

# Backing up to user .minecraft folder after completion of Minecraft
rm -r "/home/$long_name/.minecraft"
sudo \cp -r /root/.minecraft "/home/$long_name/"
sudo chown -R "$long_name:$long_name" "/home/$long_name/.minecraft"

closed 2020-10-27 by gbaman