PiNet/PiNet #90: Supporting old PC-s

dashohoxha, 2017-01-03:
In the script `pinet` the word `armhf` is used 130 times. There are also the variables `cpuArch` and `ltspBase` which are never used. If the variables were used instead, it could be easier to adopt this script for other architectures (for example i386).
If I replace `armhf` with `i386` everywhere, do you think that these scripts will work for old PCs as well?

dashohoxha, 2017-01-05:
I see that there are also referencies to Raspberry Pi repos and packages. These need to be removed or modified as well. So, it may not be so easy after all.
But I like the easy installation scripts of pinet (so that even a student can do it). I wish there were some scripts like these for i386/amd64 as well.

gbaman, 2017-01-10:
Unfortunately, there is no plans in the near future for PiNet to support any other distribution than Raspbian on any other platform than ARMHF. There are a number of reasons as to why, but the main reasons are around how heavily built around Raspbian the system is. Yes, I agree there is a lot of code that could be shared with an x86/64 setup, but separating that out is a whole other project in itself. The main issue is from the start, it was built around Raspbian and includes the hundreds of customisations made by the Raspberry Pi Foundation for Raspbian.
Right now I unfortunately don't even have enough time to work on the Pi side, let alone splitting it out to support more platforms.

The code is of course though open to be forked by anyone who wants to give it a go, I do accept pull requests!

closed 2017-01-10 by gbaman