PiNet/PiNet #93: Can't get sound to come out analog outputs when connected to DVI display

haxsaw, 2017-01-26:
I've got a small lab of 10 PiNet machines which have their video connected to displays with DVI inputs (so I'm using an HDMI to DVI converter cable), but the monitors have no sound capabilities. I have analog headphones plugged into the analog output, but I can't get any sound out of them. The RPi shows it can see the headphones when I go into the sound properties, but when I change it to output to the headphones and have a look again, it has switched back to the HDMI output. Does this require 'sudo' privilege to do (I don't have that set up on my PiNet machines), or should it be working as is? Can anyone point me to any configs where I might be able to force this change at boot time?

birving2, 2017-01-26:
Did you change the input to Analog by right clicking on the sound icon? I think the default is HDMI.

haxsaw, 2017-01-26:
Are you saying that the input has to be analog in order for the output to be analog? To be honest, I don't remember clearly if I tinkered with input, although I think I did. I'll test it again and report back here.

birving2, 2017-01-26:
Hmm. I've never seen it revert to hdmi, except on reboot. It stays on analog when you switch it for that session.

gbaman, 2017-01-26:
@haxsaw that's an interesting one I haven't stumbled across before, but I use PiNet primarily with sudo enabled. Raspbian has a stack of stuff that was just assumed by devs that the user would have root privileges, so have had plenty of stuff broken with it disabled with PiNet.
I am out of the country currently so not able to test, but might be worth enabling sudo to give it a quick test, see if it sticks? It can easily be switched off later if it makes no difference
Other thing worth trying is if Sonic Pi works with headphones selected? (in the prefs section)

To be honest, it wouldn't surprise me if switching audio without sudo did have issues given the amount of small hidden things that have cropped up on Raspbian with sudo disabled.

closed 2017-01-30 by haxsaw