PiNet/PiNet #102: PiNet Server - Rpi client login issue

JayRhithu, 2017-03-08:
Hi,
I am trying to create a PiNet Lab. After successful installation of PiNet server on Edubuntu 14.04, I couldn't login in to RPi client.

My Server details:
Edubuntu 14.04 (LTS-TrustyThar) - 32bit
AMD A8-6500 APU 
4GB RAM, 500 GBHDD.
+
Raspberry Pi Model 2 B

I got PiNet Login screen after boot. But while trying to login with username and passwordd, it shows 

'pcilib: Cannot open /proc/bus/pci
lspci: cannot find any working acess method' 

And return back to the login screen. Pls help.

[image]
[image]

gbaman, 2017-03-08:
So first off, I would suggest you may want to be looking at Ubuntu 16.04 as the PiNet server OS, as 14.04 is no longer the recommended platform (although will continue to be supported). That wouldn't be related to this issue though.

The lspci errors are not actually anything to do with this either, they simply show up behind the login screen regardless. In fact, what you are seeing is the PiNet server rejecting the login (and so the login screen restarts).

Are you trying to log in with the username used to set up the PiNet server? Try creating a new user using the PiNet software and see if you can then log in using it.

gbaman, 2017-03-17:
@JayRhithu did this solve the issue?

JayRhithu, 2017-03-25:
Hi Andrew,

First of all sorry for the delay to respond. Kindly find the attachment,
which including the details of my trials on PiNet.

Thanks & Regards,
Jayapramod

gbaman, 2017-03-25:
Certainly some very interesting results, thank you for providing the detailed documentation of the results.
PiNet really only should have required a single installation process, you shouldn't have had to do all those numerous rebuilds/reinstalls.
What has me bemused is why you are getting further sometimes than others, when running essentially the same pieces of code. This is not something I have seen reported by others much.

Your final result there still isn't right though, as you are missing the PiNet backgrounds, meaning at least 1 section of the code failed to run.

One perhaps to check, how reliable is your internet connection? If you had an unreliable connection, it would explain why you get different results each time. Unfortunately PiNet Wheezy does not work well if it does not have a stable connection throughout the entire installation process. PiNet Jessie is a little better, but because PiNet sits on top of a number of other technologies, there is not a huge amount more I can do for unreliable connections to support it catching up again when the connection is dropped.

One thing that might help is can you attach your PiNet log (found at /var/log/pinet.log) to an email and send it over to the PiNet support address (support at pinet.org.uk). It hopefully can help me dig into what is wrong with the installation.

gbaman, 2017-05-14:
As there has been nothing further on this for nearly 2 months now, am closing the issue. If it is still an issue @JayRhithu, please reopen.

closed 2017-05-14 by gbaman