PiNet/PiNet #108: Is it possible for SD boot files to hold a different raspi distribution?

danielwang405, 2017-07-07:
I understand that the PiNet boot files use raspian, but is there a way to change it to another raspberry pi os?  I'm guessing I would have to mess around with the steps on this [page](http://pinet.org.uk/articles/advanced/raspbian-kernels-process), which i know is unsupported, but is there other advice you could offer?  Thanks in advance.

gbaman, 2017-07-07:
It is not easily possible with PiNet. PiNet is specificity built around Raspbian, as it builds a full copy of Raspbian as part of its installation process. If you don't want to use Raspbian, you may have better luck interfacing with LTSP directly, which PiNet is based upon.

There is though no plans in the future for PiNet to support any other platforms other than Raspbian built from scratch (PiNet doesn't use the Pi Foundation images themselves)

closed 2017-07-07 by gbaman

danielwang405, 2017-07-07:
Thanks for the response.  I made a mistake earlier by mentioning a different raspi distribution.  I mean to say I plan on using C-Stembian[http://c-stem.ucdavis.edu/c-stembian/](url), which is built off Raspian, but with some additional programs.  Would that be more feasible?

gbaman, 2017-07-07:
PiNet doesn't work with prebuilt full Pi OS images, but in the case of c-stembian, it looks like they do have instructions for adding the required repos/packages to a base Raspbian image.
See method B - http://c-stem.ucdavis.edu/downloads/
You can chroot into the Raspbian image on the server by following the documentation here - http://pinet.org.uk/articles/advanced/editing-outside.html

Once chrooted into the image, you should just have to run the commands they include.

danielwang405, 2017-07-07:
I'll try that.  Thanks!