PiNet/PiNet #109: How to connect the hardware for PiNet

danielwang405, 2017-07-14:
Hi,
I actually have managed to install PiNet Jessie on ubuntu and flashed an sd and it boots up just fine.  My issue is that I have to be connected to wifi to start PiNet or I get this error and I am wondering if it is normal:
/bin/sh: 270: Syntax error: redirection unexpected
Once PiNet is open, the  ip shows up as 0.0.0.0 on the top of the main menu until I switch to the wired connection hooked up to the router.  Then, after checking system status, PiNet updates to the wired connection which is the ip that the raspberry pi sd's are using.  I am wondering if it is normal to be switching networks like this.   

My current setup is:  my laptop is connected to a router, not the blue ethernet port, but one of the four yellow ports (I think they're called LAN ports) .  Additionally, my raspberry pi is connected to one of those yellow ports as well.  

Additionally, I have successfully created users, shared folders, but epoptes does not show my connected pi.  Sorry, if that should be a different issue posted, but I am wondering if that is related?  Any advice is appreciated. Thanks.

gbaman, 2017-07-16:
@danielwang405 can I get the version of PiNet you are using? It is shown at the top of the PiNet software.

As for the issue with Epoptes, unfortunately this is a well known issue. The software can be a little unreliable, but is maintained by a different group of people, PiNet simply provides a nice way to add it to the installation.

danielwang405, 2017-07-16:
Hi,

 I am using PiNet 1.2.8

gbaman, 2017-07-16:
Can you also upload a screenshot of the output of the following command on the PiNet server when both wifi and wired are connected?

ifconfig

danielwang405, 2017-07-16:
I'm at home at the moment so I can't connect to the router until tomorrow. 
By PiNet server, do you mean the computer that I am using as the server?  So I would type ifconfig into a  new terminal window?

danielwang405, 2017-07-17:
This is the result of typing in ifconfig when connected to wired and wifi
[image]

closed 2018-05-19 by gbaman