PiNet/PiNet #110: Wanted - Ideas/projects to help/hack PiNet

daleq, 2017-07-20:
My company sponsors a week-long "hackathon" that starts Monday 2017-07-24 and I'm running a project called "Computer Classroom in a Crate" where all the hardware and software for a PiNet classroom will be shipped to an orphanage in Mozambique.

Depending on how many volunteers join, I may have significant resources to hack/document/test PiNet.  Are there any areas/projects that could use some resources?  Feel free to list whatever you want and we can prioritize based on numbers/expertise of volunteers.

I just discovered PiNet two weeks ago, but it implements what I've daydreamed about for quite a while.  Nice work!

ITlichutam, 2017-07-20:
Hi daleq,
we are currently working on a similar set up for Guatemala using a main server and Raspberry Pis as Thick clients. We were stuggling to use Pis of the 1st and 2nd generation, so an Idea might be to try to move for the Pis into the directin of ThinClients or Zero Clients (which posed a problem in the LTSP community because of the used prozessor if I am correctly informed).
Secondly we are running the Pis in a Virtual Machine, which causes trouble when trying to connect them to the internet over the main server (Solution: Router set up over /etc/dhcp/dhcpd.conf and forwarding IPs/ "masqerading" the Vm IP table ), this has to be done so far out of the Pinet environment, I think it would come in handy to improve the setup for Virtual machines or create a straight VM img one could fall back to. Especially in Environments like Mozambique or Guatemala a stable system and a quick replacement options are vital for functionality.
Alle the best

daleq, 2017-07-20:
Thanks ITlichutam,

I was planning on using Pi-3 hardware, so fat-clients seem reasonable.  It seems to me that given the other costs of setting up a classroom, spending a bit extra for Pi3 is worth it.

I like using VMs and have set up VMware ESXi in other locations, but I hesitated to do so on this project.  I'm struggling with "just do it simply" and "make sure it's super reliable and repairable".  :-)  I see ProxMox VE 5.0 just released, maybe that will push me over the edge to run the PiNet server in a VM.

Since I'm just starting on this project (and the classroom does not yet exist) I don't have any real-world experience so I assume there will be "bumps" on the road.

Thanks for your reply!

gbaman, 2017-07-24:
@daleq right now, there isn't a huge amount of separate large features sitting to be added for PiNet that someone else could easily do a big block of work on in one go. 
The next big one for PiNet is the move to Raspbian Stretch which I have already been working on and is nearly there.

One of the big bits though that I would love help on is the documentation. A large amount of it for is well out of date. The whole installation guide for example is actually screenshots from 2-3 years ago and could do with being completely rewritten/re-screenshotted.
If it is the type of thing you would be interested in helping out with, I can try and put together a list this evening of docs pages that would be a big help to get redone or created?

@ITlichutam unfortunately I have no intention of pushing PiNet in the direction of thin clients. Using it as a thin client turns the Pi into any other computer, without stuff like GPIO etc. Adding it as well as keeping fat clients will simply confuse other educators as to which option was for them, plus is a whole different system to maintain.
You may though want to take a look at Remote-Apps which is built into PiNet (from LTSP) and allows you to run individual apps on the server. Or, simply set up a terminal server of some sort that then auto starts a connection when the Pi starts. http://pinet.org.uk/articles/advanced/remote-apps.html
As for internet access, that is outside of PiNets scope and I believe should stay outside of it given the huge amount of different variables that can occur with networking.
As for providing a pre-built image, that is easier said than done. It would need to be constantly kept up to date and rebuilt, while also being quite large. I had done estimates of the size a while back and it came to along the lines of 12GB+ as contains Ubuntu OS and a full copy of Raspbian with a stack of other tools. This simply isn't feasible to distribute and keep up dated. For setups with less reliable internet, downloading a single 12GB file is well out of their reach, in that case individual packages which are they checked after they are downloaded makes a lot more sense.
Plus of course, the bandwidth for distributing an image that big and keeping it up to date would be rather expensive long term.
Having a single method to install and set up that generally results in the same outcome for everyone (after which they can make some simple tweaks) makes the whole system easier to maintain for myself, while also making it easier to document and then support educators using it. All unfortunately comes down to my available time.

daleq, 2017-07-24:
@gbaman 
It turns out that "Hack week" is really only two days, so limited scope is fine.  :-)  
I'll be glad to update documentation.  Let me know what pages you'd like updated (in rough order).
If you have any tips/hints, feel free to pass along.  Otherwise, I'll try to mimic image resolution, etc. from the existing pages.

gbaman, 2017-07-25:
@daleq I will start adding rewrite ideas to the following Github project. Feel free to nab any of them, especially the normal priority ones.
https://github.com/PiNet/PiNet.github.io/projects/1

The general docs contributing guide can be found here - http://pinet.org.uk/articles/CONTRIBUTING.html

As for images, in general I try to keep them under 200kb each (aiming more for like 100kb) as JPEGs.

If you have any other questions, feel free to ask!

closed 2018-05-19 by gbaman