PiNet/PiNet #112: Epoptes Not Launching On Pi

TheFlyingCrocodile, 2017-08-02:
i cant get the epoptes to run on the pi when signed in as a teacher i even modified manually /etc/group

gbaman, 2017-08-03:
@TheFlyingCrocodile when I get a free second over the next week or 2, I will try and take a look into this. Given the PiNet application is running over remote-apps, any applications spawned from it should also work.

Will take a look.

gbaman, 2017-08-03:
Scratch that, just seeing the issue is closed. @TheFlyingCrocodile did you get it working?

TheFlyingCrocodile, 2017-08-04:
@gbaman no not sure why I closed it to be honest, I guess it's because I got to to work manually with a .desktop file but still does not work via pinet menu

gbaman, 2018-05-19:
I can't seem to reproduce this one. If is still an issue, feel free to reopen.

closed 2018-05-19 by gbaman