PiNet/PiNet #114: Teacher specific sudo permissions not working

TheFlyingCrocodile, 2017-08-08:
when trying to run sudo commands when the "allow students to use sudo" option is off on a teacher account it asks for a password and does not work when the users pass is entered had the student option on and it worked fine but after its off no even chrooting into the raspbian image and edititng visudo will let me run pinet menu on the pi.

using beta 1.2.8 on pi2

gbaman, 2017-08-09:
This is actually intentional, as disabling sudo on the Raspbian chroot for students does the same for teachers.
Teachers still have sudo access on the PiNet server and should be able to launch the PiNet software from their desktop whether they have sudo access on the Pis or not.

It is something though that I hadn't given much thought to so will add to the list to investigate, adding teacher sudo access to the Raspbian OS isn't a hard thing, it just probably hasn't been raised before given nearly all educators from what I can gather, leave sudo access for students on due to the risks being limited.

TheFlyingCrocodile, 2017-08-10:
probably doesn't matter much but using beta 1.2.9 now pinet menu is still not launching on the pi as a teacher

gbaman, 2017-08-10:
Well that is weird, can you try on a Pi, in an LXTerminal running the following command?
```ltsp-remoteapps gnome-terminal```

That should launch another terminal that is actually running on the PiNet sever.
If that works, then try
```sudo bash pinet```

closed 2018-05-19 by gbaman