PiNet/PiNet #115: Unable to install Jessie version

jbartucz, 2017-08-15:
I created a fresh install of Ubuntu 16.04 on a machine and when I tried to install the Jessie alpha version, I got a bunch of errors. I have attached screen shots (sorry for the potato quality).

[image]
[image]
[image]

jbartucz, 2017-08-15:
It's not critical as I will use the Wheezy version in my classroom. Just thought you should know. Thanks!

gbaman, 2017-08-15:
Well, I would be recommending you use the Jessie version if at all possible as it is better supported by myself and the Raspberry Pi Foundation.
The Wheezy release, although generally very stable, hasn't had many updates bar security for 2 years now.

jbartucz, 2017-08-15:
Ok, I'm definitely willing to do that if I can get it to boot. The installation page is full of warnings about using it, though. Definitely willing to help you test it in any way that would be helpful. I'm not going to need my lab up and running for several weeks.

gbaman, 2017-08-15:
I probably should really remove those :)
It has been in testing since December 2016 through different stages and is near enough should be sitting in the stable channel by now, just Stretch is coming along now soon so it is getting my attention. It will be a nice in place upgrade though.

jbartucz, 2017-08-16:
Ok, so I think I made some progress in finding the error: I tried reinstalling everything and got the same error screens as above. When I try to run the commands that errored out in a terminal, I get the following error:

```
The following packages have unmet dependencies:
raspberrypi-sys-mods : Depends: systemd (>= 230) but 215-17+deb8u7 is to be installed
E: Unable to correct problems, you have held broken packages.
```

gbaman, 2017-08-16:
Can I check, are you installing PiNet on a filtered internet connection (for example in a school) or an unfiltered internet connection?

jbartucz, 2017-08-16:
Unfiltered (in my basement). Although I do have a pihole running.

gbaman, 2017-08-16:
I finally got to the bottom of the issue, was an issue with the way PiNet handles some packages. I have done some fiddling around with the PiNet apt server which should resolve it for now till I can get more time to put in place a more permanent solution.
Give it another install and see if that works now @jbartucz

jbartucz, 2017-08-17:
During the installation process, I have been selecting the "stable" release channel. Should I be selecting the "beta" or "alpha" instead?

gbaman, 2017-08-17:
Ahhh, that probably explains why I couldn't reproduce it as easily, as you had mentioned alpha in the initial issue comment so I was running with the alpha code.
Is it working after the fix yesterday evening?

jbartucz, 2017-08-17:
Makes sense - I was just following the instructions and wasn't smart enough to associate alpha version with alpha release channel. Sorry about that. 

Looks like it installed successfully. I only saw a couple minor issues. I'll close this issue.

If you want help updating the installation instructions, just send me a message.

closed 2017-08-17 by jbartucz