PiNet/PiNet #116: piclib: Cannot open /proc/bus/pci

jbartucz, 2017-08-15:
Hello, I created a new install of Ubuntu 16.04 on a machine on its own. Installed PiNet Wheezy successfully. Used the image to burn an SD card for a Raspberry Pi 3. Shows the proper login screen, but when I try to log in as any user (I created several) I get the following error:

```
pcilib: Cannot open /proc/bus/pci
lspci: Cannot find any working access method.
```

9 times and then it returns to the login screen.

The server and Pi are both plugged into a gigabit switch and the switch is connected to the LAN.

Thanks,
John

gbaman, 2017-08-15:
@jbartucz that isn't actually an error such that is important, it is just complaining it is looking for the PCI bus which of course a Pi doesn't have.
It is a relic of some old code in LTSP.
You are seeing it as that is the login system restarting after incorrect credentials (or at least it thinks they are).

Did you add the users via PiNet and have you given the PiNet server a reboot after installation?

jbartucz, 2017-08-15:
I rebooted the pi, but not the server. Hold on a sec. Be right back :)

jbartucz, 2017-08-15:
Ok, so in theory, it let me login after rebooting the server, but now I am getting hundreds of these scrolling down the screen:

```
[20231.some numbers] SQUASHFS error: Unable to read metadata cache entry [hex address]
[20231.more numbers] SQUASHFS error: Unable to read directory block [hex address:hex num]
```

gbaman, 2017-08-15:
Weird, try running an NBD recompress of the Raspbian image (option is inside the "other" menu).
Then reboot your Pi

jbartucz, 2017-08-15:
Where does it put the new image? It's not in my home folder.

gbaman, 2017-08-15:
It was rebuilding the full Raspbian image and will have replaced the old one once the image was recompressed with the new one.
(it is stored in /opt/ltsp/images)
Try booting the Pi again though now. Note that it wasn't the SD card image that was being rebuilt, so nothing needs done to the SD card you are using.

jbartucz, 2017-08-15:
Ok, now it just goes back to the login screen every time, no matter which user/password I put in. No more error messages though.

Should we just scrap this issue and move on to trying to get the Jessie installation going? If so, I'll just reinstall ubuntu from scratch on the server and put on the Jess PiNet. 

Thanks,
John

gbaman, 2017-08-15:
I would give Jessie another go. I have an install of Jessie alpha currently installing and haven't hit up against the reported issue as of yet, but the installation will still be a bit longer to go.

jbartucz, 2017-08-16:
Will do, thanks.

closed 2017-08-16 by jbartucz