PiNet/PiNet #117: Just a couple of minor issues in the Jessie install

jbartucz, 2017-08-17:
```
Collecting gpiozero
  Downloading gpiozero-1.4.0.tar.gz (102kB)
    100% |████████████████████████████████| 102kB 431kB/s 
Building wheels for collected packages: gpiozero
  Running setup.py bdist_wheel for gpiozero ... done
  Stored in directory: /home/jobartucz/.cache/pip/wheels/ab/9c/8d/3530194cb125e1c4c45fd3d178d54eb5d6349df5ae9e564ef1
Successfully built gpiozero
Installing collected packages: gpiozero
Successfully installed gpiozero-1.4.0
Collecting gpiozero
  Downloading gpiozero-1.4.0-py3-none-any.whl (97kB)
    100% |████████████████████████████████| 102kB 427kB/s 
Installing collected packages: gpiozero
Successfully installed gpiozero-1.4.0
Collecting pgzero
  Downloading pgzero-1.1.tar.gz (1.1MB)
    100% |████████████████████████████████| 1.1MB 188kB/s 
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-zFEc3_/pgzero/setup.py", line 6, in <module>
        with open(path, encoding='utf8') as f:
    TypeError: 'encoding' is an invalid keyword argument for this function

Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-zFEc3_/pgzero/
Collecting pgzero
  Using cached pgzero-1.1.tar.gz
Collecting pygame>=1.9 (from pgzero)
  Downloading pygame-1.9.3.tar.gz (3.0MB)
    100% |████████████████████████████████| 3.0MB 58kB/s 
    Complete output from command python setup.py egg_info:
    
    
    WARNING, No "Setup" File Exists, Running "config.py"
    Using UNIX configuration...
    
    /bin/sh: 1: sdl-config: not found
    /bin/sh: 1: sdl-config: not found
    /bin/sh: 1: sdl-config: not found
    
    Hunting dependencies...
    WARNING: "sdl-config" failed!
    Unable to run "sdl-config". Please make sure a development version of SDL is installed.
```

jbartucz, 2017-08-17:
```
  Found existing installation: twython 3.1.2
    DEPRECATION: Uninstalling a distutils installed project (twython) has been deprecated and will be removed in a future version. This is due to the fact that uninstalling a distutils project will only partially uninstall the project.
    Uninstalling twython-3.1.2:
      Successfully uninstalled twython-3.1.2
```

gbaman, 2018-05-19:
So these issues are to be expected as are in related to some packages simply not being available for Python 2. It can safely be ignored.

closed 2018-05-19 by gbaman