PiNet/PiNet #119: multiple 'command "sudo ltsp-chroot....." failed to execute correctly with a return code of 100' errors.

danielwang405, 2017-08-25:
Hi there,
I keep running into 'Command failed to execute' errors for different packages like i2c tools, python digitalio like in the screenshot below. While I have pressed 'Continue', there have been over a dozen now so I'm afraid that there are more serious issues.

[image]

I am currently running Ubuntu 16.04 using Wubi for dual boot.  I am trying to install PiNet Jessie Beta.
I have tried entering
```
sudo apt-get install ubuntu-restricted-extras
sudo apt-get update && sudo apt-get upgrade
```
prior to installing PiNet, but I still run into this.

If it helps, I took a screenshot between the purple error screens to see what the terminal was outputting.
[image]
[image]

Any help/advice is appreciated!
Thanks in advance.

gbaman, 2017-08-25:
To check, are you running your PiNet server behind any sort of web filtering? Or is this installation taking place behind a corporate/school filtered web connection?

danielwang405, 2017-08-25:
Yes, the installation is taking place behind a school filtered web connection.  It's a bit strange though, because I have successfully installed PiNet the same way on the same connection in the past without bumping in those messages and on my home network, which shouldn't be filtered, I have ran into these issues.

gbaman, 2017-08-25:
The errors are suggesting it can't reach the archive.raspberrypi.org site, which is the main Raspberry Pi Foundation maintained repository for packages. Unfortunately that is out of the scope of PiNet itself at that stage. The fact it can't resolve archive.raspberrypi.org suggests either your DNS server is having some issues, their DNS server is having some issues, or something is blocking the traffic in the middle.
Can you try on a web browser from the PiNet server seeing if you can access it? I can confirm I can currently

danielwang405, 2017-08-28:
I think you were right about the web filtering possibly causing it.  I switched from the lab/office connection to the school wide network and I stopped running into these errors.
Thanks!

closed 2017-08-28 by danielwang405

KwesiAsamoah, 2018-02-12:
I was having the same problem, i solved it by adding Raspbian Repository to my sources list

-  `sudo nano /etc/apt/sources.list` 

- add this to the end of the file 
```
 deb http://archive.raspbian.org/raspbian wheezy main contrib non-free
 deb-src http://archive.raspbian.org/raspbian wheezy main contrib non-free
 ```
-   hit `ctrl + x` then `y` then `enter` to exit out of nano

If needed, you can store the Raspbian public key into your apt-get keyring using the following command: 
`wget https://archive.raspbian.org/raspbian.public.key -O - | sudo apt-key add -`

gbaman, 2018-02-12:
@KwesiAsamoah PiNet does everything mentioned above?

Adding Raspbian repository - https://github.com/PiNet/PiNet/blob/jessie-stable/pinet#L344
Adding key - https://github.com/PiNet/PiNet/blob/jessie-stable/pinet#L201

Do note, that as is generally best practice, the apt repo is added by PiNet to a separate file in /etc/apt/sources.d, as opposed to directly in the main sources.list file.

So there will be another underlying issue with your web connection (or may have been when you first tried), as was the case for the other user in this issue.

If you instead mean you added it to the main Ubuntu OS, this is a **very bad idea**. The packages are armhf compiled packages! They are not compatible with x86 machines and will cause a large number of issues later on.

KwesiAsamoah, 2018-02-13:
Thank you for your advice. I added it the **main Ubuntu OS sources list**. I did after remove it once the installation process was done for as you rightly would cause problems down the line, thanks again.