PiNet/PiNet #126: Using original boot files

danielwang405, 2017-09-15:
Hi, 
I want to undo changes I've made in chroot after editing the ltsp image and recompressing.  I've downloaded the original boot files from the PiNet/Boot repo and placed them into the PiBoot folder, but now I'm unsure how to get PiNet to recognize this change.  I've tried starting PiNet, but the raspberry pi won't boot at all.

Thanks in advance.

TheFlyingCrocodile, 2017-10-02:
from the main pinet menu go to rebuild OS and it will reinstall just the image from scratch keeping your users and data (any programs will need to be reinstalled though)

gbaman, 2017-10-03:
@danielwang405 what were these changes you made? As @TheFlyingCrocodile notes, one of the options is simply rebuilding the full Raspbian copy chroot with the Rebuild OS option. This will take a few hours, but does mean you start with a fresh Raspbian chroot ready to go, but do loose any customisations you have made.

danielwang405, 2017-10-11:
I see now.  I'm using chroot to install an OS that is a superset of raspbian.  Sometimes there are issues and I wasn't sure how to restore the originals, so I instead reinstalled PiNet which meant reinstalling Ubuntu.  I can't believe I didn't think of rebuilding.  Thanks

closed 2017-10-11 by danielwang405