PiNet/PiNet #127: Prevent/Change overwrite of config.txt

robp2175, 2017-09-28:
Having to complete rebuild the image to make a change to config.txt is kind of a pain. The fact is that I may need to use a different config.txt for different machines. How can I prevent the config.txt from being overwritten?

gbaman, 2017-09-28:
The idea behind the config.txt file edits are that in most cases, they only need to be made once in a blue moon. They piggy back off the same mechanism used for kernel updates which sit inside the main image, so will continue to need a rebuild of that image each change.

If you want to support multiple different hardware setups using the same PiNet server, each requiring different config.txt file values, then your answer is conditional filtering which is part of the Raspberry Pi config.txt spec and is completely unrelated to PiNet itself.

You can filter values by monitor ID, Raspberry Pi serial number and types of Pi for example - https://www.raspberrypi.org/documentation/configuration/config-txt/conditional.md

robp2175, 2017-09-28:
Thank You Andrew, Easy solution to my problem. I was clearly overcomplicating it. I appreciate you pointing me in the right direction.

gbaman, 2017-09-29:
Great, glad I could help.

closed 2017-09-29 by gbaman