PiNet/PiNet #128: RASPBIAN DESKTOP

TheFlyingCrocodile, 2017-10-02:
now that the rpi foundation has the raspberry pi desktop image to use raspbian on amd65 hardware why not add it as a second chroot for teachers who are using desktops to manage the pis in a class

gbaman, 2017-10-03:
The main reason for not going near the x86 port is there are a number if differences under the hood that would result in 2 actual different build processes, at least partially anyway. As well, right now PiNet is heavily hard coded for using armhf, changing that would be a task in itself.

It is certainly something I would hope to be able to investigate further in the future, but Stretch is first on the todo list for now.

gbaman, 2018-05-19:
Given the scope of PiNet already is pretty wide, I have decided that am not going to go ahead with this one anytime soon.
Just maintaining the armhf Raspbian is enough of a workload for now...

closed 2018-05-19 by gbaman