PiNet/PiNet #130: libraspberrypi

TheFlyingCrocodile, 2017-10-06:
i changed from beta to stable and now all the libraspberrypi... packages are refusing to install and are on hold due to being broken
"Some packages could not be installed. This may mean that you have requested an impossible situation or if you are using the unstable distribution that some required packages have not yet been created or been moved out of incoming." and now none of the usb devices plugged into the pis are not working in the OS it detects them fine when running the LTSP script but when it comes to using them nada

gbaman, 2017-10-10:
@TheFlyingCrocodile does it say anywhere which packages it is looking for?
Also, try giving ```sudo ltsp-chroot --arch armhf apt-get install -f``` a go, as that will get it attempting to fix broken packages.

TheFlyingCrocodile, 2017-10-11:
@gbaman yep tried chrooting in an running the fix but it just says as above and the packages its looking for are a bunch of libraspberrypi(insert here) packages

gbaman, 2017-10-11:
@TheFlyingCrocodile does it give a version number it is trying to find for libraspberrypi?

TheFlyingCrocodile, 2017-10-11:
@gbaman no its like libraspberrypi-dev etc im at home currently so i cant check

gbaman, 2017-10-31:
@TheFlyingCrocodile I am having difficulty reproducing this one.
Can you upload the output of
```ls /opt/ltsp/armhf/lib/modules```
 please?

closed 2017-12-02 by gbaman