PiNet/PiNet #135: Modify PiBoot to reboot if unable to reach server

robp2175, 2017-11-29:
So i have quite a few Pi clients that I have modified to run as webkiosks. We have lost power once or twice and the clients have come back up before the server and failed. Could someone please guide me to how I might be able to have the clients automatically reboot if they are unable to find the server?

Any help is greatly appreciated.

gbaman, 2017-12-02:
@robp2175 an interesting one and one right now without an answer unfortunately. That bit is handled by LTSP well below PiNet, in the initramfs on the SD card. I will add it as something to look into, but given the current todo list, may be a while before it is investigated.

bauman, 2017-12-03:
@robp2175 I had the same problem

LTSP correctly throws a kernel panic if NBD doesn't mount. You just need to update the grub line to deal with it.

Add panic=N to the LTSP grub line where N is the number of seconds you want to wait before a reboot.

N=1 reboots quick.  Assuming the server, clients and network gear all get power at the same time, I set my N at 120 seconds 

It's this file you need to change.  
https://github.com/PiNet/PiNet-Boot/blob/master/boot/cmdline.txt

Every time your pinet app updates your SD image, you'll have to add the panic arg yourself.

Here's my full cmdline.txt.  

`dwc_otg.lpm_enable=0 console=serial0,115200 kgdboc=serial0,115200 console=tty1 init=/sbin/init-ltsp nbdroot=1.1.1.1:/opt/ltsp/armhf root=/dev/nbd0 elevator=deadline rootwait panic=120 `

@gbaman rebooting might not be a desirable default, but it would be super if there was an option in the "update SD" menu.

closed 2020-10-27 by gbaman