PiNet/PiNet #136: sudo ltfs-chroot commands giving error code 126.

KlwntSingh, 2017-12-08:
Hi There,
I am trying to install PINet on Ubuntu xenial distro and couple of **ltfs-chroot** commands are failing giving return code 126 like 
`Command "['sudo', 'ltsp-chroot', '--arch', 'armhf', 'groupadd', 'gpio', '-g', '625']" failed to execute correctly with a return code of 126. Would you like to continue and ignore the error or retry the command?`

I have attached the screen of errors and of Ubuntu distro details.

[image]

[image]

[image]

[image]

[image]

Thanks in Advance!

KlwntSingh, 2017-12-13:
Please Help related to above issue.

gbaman, 2017-12-13:
Apologies, I missed this one coming in. Can I check the first one, is this PiNet server being installed being a school firewall or otherwise limited (or unreliable) internet connection?

Also, is the Ubuntu installation a fresh clean install with no additional software and a single user account (besides root)?

KlwntSingh, 2017-12-13:
Hi There,
Thanks for replying.
I am installing PiNet server in my home environment and also internet connection is reliable.
Yes, Ubuntu is fresh clean install, Actually i am installing PiNet server inside of lxc-container of ubuntu xenial version.

Thanks

gbaman, 2017-12-13:
Unfortunately I am unable to provide support for PiNet running inside a containerised environment given so many other things are then at play.
The only suggestion I would have to help better understand what is going on is if you can get a screenshot of the full error message of one of the commands.
For example
``` sudo ltsp-chroot --arch armhf groupadd i2c -g 627```

KlwntSingh, 2017-12-14:
Hi,
Already Attached screenshots is the only error messages I am getting.
I will try to run PiNet outside container than, it could be the reason.

Thanks for help!

gbaman, 2017-12-16:
@KlwntSingh the PiNet logs (which you attached) don't capture the full output/error message of the command, hence why I need you to if possible try running the suggested command directly in a terminal, to capture the full output.
The logs unfortunately only capture the command being run and the error code it failed with, which isn't useful in this case.

KlwntSingh, 2017-12-16:
Hi @gbaman 
Definately I will check and and will respond you back with logs.

Thanks

gbaman, 2018-01-17:
@KlwntSingh Given no response in a month now, am closing this issue. Please re-open if is still an issue.

closed 2018-01-17 by gbaman