PiNet/PiNet #137: Edublocks package?

birving2, 2018-01-17:
I'd love to install Edublocks on my PiNet server but so far have not been able to. Has anyone had any success?

gbaman, 2018-01-17:
The plan for Edublocks support with PiNet has always been to include it once the long awaited Raspbian package is released (assuming is released for Jessie/Stretch). 
Last I checked the main install script might cause some installations on non standard Raspbian installations, like PiNet.
@JoshuaLowe1002 any progress on a Debian/Raspbian package?

gbaman, 2018-03-22:
Poke @JoshuaLowe1002, any updates?

JoshuaLowe1002, 2018-11-02:
Hi yes, this should work now!

JoshuaLowe1002, 2018-11-02:
Sorry i'm really late on this, but there is no reason it should not work now @birving2

gbaman, 2018-11-02:
@JoshuaLowe1002 Great to hear, although I am not seeing an edublocks package in the public Raspbian/RPF apt repos or on your website? Am I looking in the wrong place?

JoshuaLowe1002, 2018-11-03:
We haven't got to the debian package stage purely because of this > https://github.com/AllAboutCode/EduBlocks/issues/26#issuecomment-434043481 Once this update is done I plan on packaging the link package up which should be a really easy thing to do now. However, i don't see a reason why the installer which has been updated since this issue was opened should not work with PiNet.

gbaman, 2018-11-11:
@JoshuaLowe1002 It is possible - http://pinet.org.uk/articles/advanced/editing-outside.html
But, the key thing about this is it is not maintainable long term, hence why I don't integrate any software anymore that isn't installed via a package manager (e.g. Apt or Pip).

Otherwise, it becomes very difficult to handles updates for example long term.

JoshuaLowe1002, 2018-11-14:
Understood. 

Once the new EduBlocks revision is out, it should be much easier to package up.

JoshuaLowe1002, 2018-11-14:
Probably will actually be a Pip install

closed 2020-10-27 by gbaman