PiNet/PiNet #142: PiNet Stretch status

gbaman, 2018-03-06:
Given I have had a few questions over the past while on "Where is PiNet Stretch?" and "Why does PiNet still used Jessie instead of Stretch?", this issue is here to include the status of the PiNet Stretch release.

So far, the main code for PiNet Stretch is complete and can be found over on the [Stretch-Alpha](https://github.com/PiNet/PiNet/tree/stretch-alpha) branch. 
There are though a few parts left
- The main bug left is in some random cases, [LDM](http://manpages.ubuntu.com/manpages/trusty/man1/ldm.1.html), which handles the login screen seems to crash and leave a black and unresponsive screen up. In most cases this is resolved with a reboot, but it is too frequent current to allow the PiNet Stretch release to go ahead.
- The migration code is also not yet complete. Although the migration between PiNet Jessie -> PiNet Stretch should be a lot easier than the PiNet Wheezy -> PiNet Jessie process given the upgrade can be done without deleting the OS, this requires extensive further testing. Initially I am leaning towards only allowing the chroot to be rebuilt completely (like Wheezy->Jessie).

Unfortunately all of the above is simply taking much longer than expected due to my time being far more limited now than it was a few years back.

I will keep this Github issue up to date with the PiNet Stretch status as more gets completed.

## Testing
Want to test out PiNet Stretch? The install process isn't that much different to normal PiNet. Note this must be done on a completely fresh install of Ubuntu 16.04. Migration is not yet complete.
1. ```wget https://raw.githubusercontent.com/PiNet/PiNet/stretch-alpha/pinet```
2. ```wget https://raw.githubusercontent.com/PiNet/PiNet/jessie-stable/Scripts/pinet_functions_python.py```
3. ```sudo mv pinet_functions_python.py /usr/local/bin/```  
4. ```echo "ReleaseChannel=alpha" | sudo tee /etc/pinet```
5. ```sudo bash pinet```

Feel free to post any feedback on the process below.

mcolliver, 2018-06-21:
Having tried the instructions above I think line 2 needs to be (as this now seems to work for me):

`wget https://raw.githubusercontent.com/PiNet/PiNet/stretch-alpha/Scripts/pinet_functions_python.py`

gbaman, 2018-06-21:
Whoops, good catch @mcolliver, it was needed to still be pulled from the Jessie branch initially, but now the Python code is caught up, the Stretch version is needed as well. 
Have now changed, thanks.

gbaman, 2018-08-24:
I can now confirm another bug has been fixed. This case, was related to the file manager (PCManFM).
Raspberry Pi Trading switched over (using a raspberrypi-ui-mods update) to "simplified" mode for the file manager. Although it simplified the UI a little, it also removed the "Places" sidebar, replacing it will a full system tree.
This made accessing removable drives (like micro:bits etc) very difficult and shared folders as well.

Pictures - https://twitter.com/gbaman1/status/1026459946727206913

Fixed so now sets the config option back.

https://github.com/PiNet/PiNet/commit/2de679e839e715e985371084c3d61a32bd3f2162

Still plenty more bugs unfortunately though blocking PiNet Stretch.

bbinet, 2018-08-25:
@gbaman thanks for the status update.
Could you share the list of existing stretch bugs you are aware of?

gbaman, 2018-08-25:
@bbinet I have updated the initial comment with some of the bugs I am aware of currently, but there is likely to be others missing that I have yet to come across.

gbaman, 2018-08-25:
And there is another issue fixed, the slow startup issue was related to LTSP trying to fetch an lts.conf from tftp when the tftp server wasn't actually running (or generally used with PiNet).
Now fixed in https://github.com/PiNet/PiNet/commit/7fde8a803f2d249d9d0a7e1909d4096307069ca7.

bbinet, 2018-08-28:
@gbaman thanks for the updated list of bugs remaining in stretch, this is good to know before switching.

gbaman, 2019-02-24:
Fixed a few bugs this weekend including
- raspberrypi-ui-mods overwriting PiNet configs (desktop backgrounds for example).
- Unable to launch PiNet from teacher accounts on the Pi.
- A few Ubuntu 18.04 bugs.

At this stage, mainly just leaves the new boot menu code integrating into the kernels build process, fixing Sonic Pi startup times which will probably need advice from @samaaron, (potential, but unreproducable audio playback issues) and then testing of the migration system.

closed 2020-10-27 by gbaman