PiNet/PiNet #143: Scratch 2.0 Image for Raspberry Pi

fearnma, 2018-03-16:
Is there anyway of updating on the pi software to Scratch 2.0 ?

Thanks

Matt

gbaman, 2018-03-17:
Yip, from the PiNet software on the PiNet server, hit install program and put in `scratch2`.
I apparently forgot to include it in the install process, but in general from conversations with educators, most folk seem to find it too slow to use anyway.

fearnma, 2018-05-10:
If i have had to reimage a SD card on a pi, how do i push out from the server additional software to it like scratch2?  What option do i pick on the server?

gbaman, 2018-05-19:
This is automatic, as long as it is installed by the PiNet itself (into the chroot), then it should just be on the Pis.
It won't though create a desktop icon, you may need to open a terminal on the Pi and enter ```scratch2```.

closed 2018-05-19 by gbaman