PiNet/PiNet #144: Pushing out new installed software to raspberry pi on pinet

fearnma, 2018-05-10:
If i have had to reimage a SD card on a pi, how do i push out from the server additional software to it like scratch2? What option do i pick on the server?

gbaman, 2018-05-10:
The image on the SD card is simply enough to get the Pi started up and connected to the main PiNet server, where it mounts the main filesystem (with all the software etc).
So re-imaging the cards changes nothing about the main filesystem software. You can re-image the SD cards at any time without any worries of that changing the applications etc.

fearnma, 2018-05-11:
thanks, on my main teachers pi it doesn't seem to be showing the scratch2 application in the menu.  It will hopefully sort itself out when rebooted.

gbaman, 2018-05-13:
So it is accessible on some Pis on not others? You will probably need to make sure scratch 2 is installed by putting scratch2 into the additional software option in the main PiNet menu.
Do keep in mind if you install it on a Pi, that will be cleared on a reboot as it simply then boots the image off the PiNet server. Only files you create/edit in your home folder will persist across reboots.

gbaman, 2018-05-19:
Given no further up after a week, closing this issue.
@fearnma feel free to open it if the issue is still a problem.

closed 2018-05-19 by gbaman