PiNet/PiNet #146: Pinet on Raspberry Pi 3B+ Model

fearnma, 2018-05-23:
I have just purchased one of the latest B+ models and want to run it on the PINET network within school however it will not load past the rainbow screen.  Any ideas of what the problem could be?

Thanks

Matt

birving2, 2018-05-23:
I asked Andrew about this a week ago, and he said:

"As for 3B+, that needs to be running PiNet Alpha. I need to find the time to properly finish 3B+ support as is still some issues with hardware accelerated stuff (mainly Pi camera)"

fearnma, 2018-06-21:
Any news Andrew on when pinet will be up and running for Pi3B+?

gbaman, 2018-06-21:
@fearnma as mentioned by @birving2, PiNet Jessie Alpha currently does support the Raspberry Pi 3B+.
Until PiNet Stretch is ready, that is probably unfortunately the level Pi3B+ support will sit at, Alpha, as 3B+ support was backported by hand my me to Jessie, as the Raspberry Pi Foundation themselves didn't provide it as an option. The result is there are some known bugs caused, especially aren't hardware accelerated software.
PiNet Stretch (which includes native 3B+ support) is still causing some issues (more related to it seems some bugs in LTSP Jessie), but my hope is still to have it ready mid way through the summer for educators with plenty of time for September.
You are able to give PiNet Stretch Alpha (with native 3B+) support a go on a fresh install. Although I think I have sorted all the main bugs, there are still 1 or 2 more minor ones plus heavy testing needed before a proper release (plus then the migration tools). https://github.com/PiNet/PiNet/issues/142

closed 2020-10-27 by gbaman