PiNet/PiNet #147: ltsp-chroot fails because packages are unauthenticated

skspurling, 2018-05-29:
Installing stretch-alpha, and I keeps getting error messages saying the ltsp-chroot software install fails because apt-get isn't getting --allow-unauthenticated on the apt-get command line when installing scratch and the pygame packages.

gbaman, 2018-05-29:
Can I check, are you trying to install PiNet using a filtered internet connection? 
What you are describing is common when a web filter/proxy is blocking parts of Raspbian repository used to build the PiNet installation.

skspurling, 2018-05-29:
I have a PiHole server at home, and found that one of the blacklists was impacting raw.githubusercontent.com. I white listed that. I attempted the install on two separate networks (One was my home network with the PiHole server, where I do updates and installs on my Pi’s all the time), and one I am sure doesn’t have any proxy or filter.

Shannon Spurling

shannon@more.net

gbaman, 2018-05-30:
So thing is, you really shouldn't be running apt with ```--allow-unauthenticated``` as that means it doesn't verify the package signatures.

Also, I just noticed you mentioned you are running the stretch-alpha branch. As noted over on the [PiNet releases page](http://pinet.org.uk/articles/installation/versions.html), I am not formally supporting the Stretch PiNet branches yet as there are still plenty of bugs in them (joys of doing this all in my rather limited free time...)
I would suggest sticking to the Jessie branches for now as they are much more extensively tested. If this is an issue with the Stretch branch, I will take a look at it when back working on Stretch for PiNet (it isn't top of the todo list currently)

closed 2018-08-21 by gbaman