PiNet/PiNet #150: Desktop link issues

gldneagl, 2018-07-17:
I have tried several time to install Pinet on virgin builds of both Ubuntu 16 and Ubuntu-Mate 16 desktop versions so I tried the Ubuntu 16 server version and ended up with this same issue... desktop links that do not work after what seems otherwise to be a successful full install.

[image]
.

gbaman, 2018-08-21:
@gldneagl the desktop links should not work on the PiNet server. They are actually desktop links for use on the Raspberry Pis. They just carry over to the server as well as the same account can be used in both places.

gbaman, 2018-09-04:
@gldneagl It certainly would be tidier yes, but I am not aware of a way to hide them without it also hiding on the Pi. The Desktop folder is part of the users home folder, so is the same no matter where you log in unfortunately.

closed 2018-09-04 by gbaman

gldneagl, 2018-09-06:
I am going to investigate, since having them in the folder I created for them on the server desktop meant that the update SD card scripts couldn't find them. I moved them back out of the folder, and voila!, the update went without an error about not finding pinet-password.desktop