PiNet/PiNet #151: PiNet Stretch Alpha Audio Issues

ferrabacus, 2018-08-21:
Running PiNet Stretch Alpha with a class size of 32 in a virtual machine on Ubuntu 16.04 and we cannot access audio at all. 

I have to use the PiNet Stretch Alpha because it is the only version I could get working with the Raspberry Pi 3B+. I made a virtual machine with stretch in beta & stable and jessie stable; none of them work with the 3B+. 

Attempts to fix the problem:

1. Use bluetooth instead of headphone jack. (Didn't work)
2. Right clicked audio button in toolbar and manually set it to analog (Didn't work)
3. Sudo Raspi-Config and forced audio to go through 3.5mm jack (Didn't work).
4. Try all three options above with different headsets. (Didn't work). 

Any advice on how to fix this issue?

gbaman, 2018-08-21:
@ferrabacus audop issues is certainly an interesting one. PiNet Jessie Alpha should be working fine for audio output on Pi 3B+ (we use it all the time at events).
PiNet Jessie Alpha does certainly have a few bugs given it was basically a hotfix to get Pi 3B+ support working while work continued on PiNet Stretch (which is not yet ready), but audio is something that should be working fine there.

You do though mention you are instead running PiNet Stretch Alpha, which isn't ready for prime time just yet. I wouldn't advise running it in a classroom setting at this stage. Best bet would be do a reinstall using PiNet Jessie Alpha for now, that should fix the audio issues.

Before doing that though, is there any way you could also test with an HDMI monitor that includes speakers? Just to check if audio output is broke full stop, or if the switching is broke?
Finally, what are you using to create the audio being played?

ferrabacus, 2018-08-22:
@gbaman 

I'll make the switch to Jessie Alpha soon when I dual boot the host machine instead of using a virtual machine. The host machine has great specs, but it just seems that the virtual machine becomes extremely laggy even with 2 CPUs, 16gb ram, and solid state dedicated to the virtual machine; none of which exceeded even half the host machine. Unless you have any suggestions here?

I will try testing this with a monitor with audio, but currently we don't have one set up with built in speakers. I'll have this tested by the end of the week. 

We used many different types of audio sources, youtube mostly, but they all came from the browser.

A student and I created a quick fix for the meantime. We created a shell script with two lines of code that students can execute each time they log in that gets audio working. This isn't an ideal fix, but it'll work until we change over. 

This is what is inside the script
`sudo modprobe snd_bcm2835`
`sudo amixer cset numid=3 1`

closed 2019-04-04 by ferrabacus