PiNet/PiNet #155: Epoptes permission denied

squadramunter, 2019-04-04:
An error occurred while trying to connect to the epoptes service: Permission denied

User pinet must be a member of group teacher to run epoptes.

What am I doing wrong? I add user pinet to teacher group but nothing works?

Do I need to restart the system because logout did not work for me.

All help is grateful

gbaman, 2019-04-04:
@squadramunter in general for groups stuff, I would advise giving the PiNet server a reboot yes. This is especially the case for a fresh installation.

squadramunter, 2019-04-04:
Thanks this did the trick! I hope the documentation can be updated soon!
http://pinet.org.uk/articles/guides.html#epoptes_classroom_management_software_documentation_not_complete_yet

closed 2019-04-04 by squadramunter